project-chip/connectedhomeip #14777: [TC-TM-1.1]- In Temperature Measurement Cluster, the cluster revision value is not matched with the expected outcome.

Rajashreekalmane, 2022-02-04:
In Temperature Measurement Cluster, testcase 1.1 (TC-TM-1.1), the "cluster revision attribute" value must be 4 as per the Testplan,
But the actual outcome is 3. This needs to be fixed as per Test plan expected outcome.

Test Test-plan link:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/temperaturemeasurement.adoc

expected behavior - cluster revision attribute - 4
actual behavior - cluster revision - 2

Actual command:- ./chip-tool temperaturemeasurement read cluster-revision 1 1

Actual log is attached below
[[TC-TM-1.1]-Cntr.txt](https://github.com/project-chip/connectedhomeip/files/8001077/TC-TM-1.1.-Cntr.txt)

cjandhyala, 2022-02-04:
dup to https://github.com/project-chip/connectedhomeip/issues/12190

woody-apple, 2022-02-08:
https://github.com/project-chip/connectedhomeip/issues/12190

closed 2022-02-08 by woody-apple