project-chip/connectedhomeip #14786: [TC-TM-2.1][TC-TM-2.2] Chip-tool: failed to process Measured Value attribure read response.

AntonGrey, 2022-02-04:
#### Temperature Measurement cluster: read Measured Value attribute with chip-tool or TestHarness.
 * Expected behavior: success read Measured Value.
 * Actual behavior: failure response, but ReportDataMessage has success response with right Measured Value.
 * Reproduce command: `./chip-tool temperaturemeasurement read measured-value 12344321 1`

**Chip-tool log:**
```
[1643973317.845061][1464486:1464491] CHIP:EM: Removed CHIP MessageCounter:15506481 from RetransTable on exchange 49929i
[1643973317.845099][1464486:1464491] CHIP:DMG: ReportDataMessage =
[1643973317.845112][1464486:1464491] CHIP:DMG: {
[1643973317.845119][1464486:1464491] CHIP:DMG:  AttributeReportIBs =
[1643973317.845134][1464486:1464491] CHIP:DMG:  [
[1643973317.845144][1464486:1464491] CHIP:DMG:          AttributeReportIB =
[1643973317.845160][1464486:1464491] CHIP:DMG:          {
[1643973317.845170][1464486:1464491] CHIP:DMG:                  AttributeDataIB =
[1643973317.845179][1464486:1464491] CHIP:DMG:                  {
[1643973317.845193][1464486:1464491] CHIP:DMG:                          DataVersion = 0x0,
[1643973317.845204][1464486:1464491] CHIP:DMG:                          AttributePathIB =
[1643973317.845214][1464486:1464491] CHIP:DMG:                          {
[1643973317.845231][1464486:1464491] CHIP:DMG:                                  Endpoint = 0x1,
[1643973317.845241][1464486:1464491] CHIP:DMG:                                  Cluster = 0x402,
[1643973317.845255][1464486:1464491] CHIP:DMG:                                  Attribute = 0x0000_0000,
[1643973317.845267][1464486:1464491] CHIP:DMG:                          }
[1643973317.845282][1464486:1464491] CHIP:DMG: 
[1643973317.845295][1464486:1464491] CHIP:DMG:                                  Data = 25, 
[1643973317.845306][1464486:1464491] CHIP:DMG:                  },
[1643973317.845321][1464486:1464491] CHIP:DMG: 
[1643973317.845331][1464486:1464491] CHIP:DMG:          },
[1643973317.845347][1464486:1464491] CHIP:DMG: 
[1643973317.845356][1464486:1464491] CHIP:DMG:  ],
[1643973317.845372][1464486:1464491] CHIP:DMG: 
[1643973317.845382][1464486:1464491] CHIP:DMG:  SuppressResponse = true, 
[1643973317.845391][1464486:1464491] CHIP:DMG: }
[1643973317.845488][1464486:1464491] CHIP:TOO: Default Failure Response: 0x01
[1643973317.845528][1464486:1464491] CHIP:EM: Sending Standalone Ack for MessageCounter:10718731 on exchange 49929i
[1643973317.845562][1464486:1464491] CHIP:IN: Prepared encrypted message 0x7f8f4d1a6070 to 0x0000000000BC5C01 (1)  of type 0x10 and protocolId (0, 0) on exchange 49929i with MessageCounter:15506482.
[1643973317.845583][1464486:1464491] CHIP:IN: Sending encrypted msg 0x7f8f4d1a6070 with MessageCounter:15506482 to 0x0000000000BC5C01 (1) at monotonic time: 97913460 msec
[1643973317.845644][1464486:1464491] CHIP:EM: Flushed pending ack for MessageCounter:10718731 on exchange 49929i
[1643973317.845770][1464486:1464486] CHIP:-: examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error at examples/chip-tool/commands/common/CHIPCommand.cpp:59
[1643973317.845814][1464486:1464486] CHIP:TOO: Run command failure: examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error
[1643973317.848526][1464486:1464486] CHIP:SPT: VerifyOrDie failure at examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)
```

**TestHarness log:**
```
INFO     | 2022-02-04 11:14:17.066 | Executing Test Step: read the mandatory attribute: MeasuredValue
CHIPTOOL | 2022-02-04 11:14:17.067 | [1643973251.060004][65:75] CHIP:TOO:  ***** Test Step 1 : read the mandatory attribute: MeasuredValue
CHIPTOOL | 2022-02-04 11:14:17.173 | [1643973251.060273][65:75] CHIP:DMG: SendReadRequest ReadClient[0xffff94001980]: Sending Read Request
CHIPTOOL | 2022-02-04 11:14:17.175 | [1643973251.060507][65:75] CHIP:IN: Prepared encrypted message 0xaaaab6a1b960 to 0x69A040DD86EFBB53 (1)  of type 0x2 and protocolId (0, 1) on exchange 35898i with MessageCounter:4940070.
CHIPTOOL | 2022-02-04 11:14:17.177 | [1643973251.060577][65:75] CHIP:IN: Sending encrypted msg 0xaaaab6a1b960 with MessageCounter:4940070 to 0x69A040DD86EFBB53 (1) at monotonic time: 10243673 msec
CHIPTOOL | 2022-02-04 11:14:17.292 | [1643973251.060801][65:75] CHIP:DMG: MoveToState ReadClient[0xffff94001980]: Moving to [AwaitingIn]
CHIPTOOL | 2022-02-04 11:14:17.293 | [1643973251.060883][65:75] CHIP:CTL: Device connected callback with null pairing delegate. Ignoring
CHIPTOOL | 2022-02-04 11:14:17.295 | [1643973251.060930][65:75] CHIP:EM: Sending Standalone Ack for MessageCounter:1910341766 on exchange 35897i
CHIPTOOL | 2022-02-04 11:14:17.296 | [1643973251.060982][65:75] CHIP:IN: Prepared plaintext message 0xffff9bffd9c8 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 35897i with MessageCounter:1440468152.
CHIPTOOL | 2022-02-04 11:14:17.298 | [1643973251.061033][65:75] CHIP:IN: Sending plaintext msg 0xffff9bffd9c8 with MessageCounter:1440468152 to 0x0000000000000000 at monotonic time: 10243674 msec
CHIPTOOL | 2022-02-04 11:14:17.299 | [1643973251.061164][65:75] CHIP:EM: Flushed pending ack for MessageCounter:1910341766 on exchange 35897i
CHIPTOOL | 2022-02-04 11:14:17.408 | [1643973251.067117][65:75] CHIP:EM: Received message of type 0x5 with protocolId (0, 1) and MessageCounter:13013002 on exchange 35898i
CHIPTOOL | 2022-02-04 11:14:17.410 | [1643973251.067178][65:75] CHIP:EM: Found matching exchange: 35898i, Delegate: 0xffff94001980
CHIPTOOL | 2022-02-04 11:14:17.411 | [1643973251.067229][65:75] CHIP:EM: Rxd Ack; Removing MessageCounter:4940070 from Retrans Table on exchange 35898i
CHIPTOOL | 2022-02-04 11:14:17.413 | [1643973251.067261][65:75] CHIP:EM: Removed CHIP MessageCounter:4940070 from RetransTable on exchange 35898i
CHIPTOOL | 2022-02-04 11:14:17.524 | [1643973251.067339][65:75] CHIP:DMG: ReportDataMessage =
CHIPTOOL | 2022-02-04 11:14:17.525 | [1643973251.067381][65:75] CHIP:DMG: {
CHIPTOOL | 2022-02-04 11:14:17.527 | [1643973251.067409][65:75] CHIP:DMG: 	AttributeReportIBs =
CHIPTOOL | 2022-02-04 11:14:17.528 | [1643973251.067456][65:75] CHIP:DMG: 	[
CHIPTOOL | 2022-02-04 11:14:17.531 | [1643973251.067486][65:75] CHIP:DMG: 		AttributeReportIB =
CHIPTOOL | 2022-02-04 11:14:17.637 | [1643973251.067527][65:75] CHIP:DMG: 		{
CHIPTOOL | 2022-02-04 11:14:17.638 | [1643973251.067556][65:75] CHIP:DMG: 			AttributeDataIB =
CHIPTOOL | 2022-02-04 11:14:17.640 | [1643973251.067594][65:75] CHIP:DMG: 			{
CHIPTOOL | 2022-02-04 11:14:17.641 | [1643973251.067627][65:75] CHIP:DMG: 				DataVersion = 0x0,
[... 4 lines cut ...]
CHIPTOOL | 2022-02-04 11:14:17.865 | [1643973251.067786][65:75] CHIP:DMG: 					Attribute = 0x0000_0000,
CHIPTOOL | 2022-02-04 11:14:17.866 | [1643973251.067815][65:75] CHIP:DMG: 				}
CHIPTOOL | 2022-02-04 11:14:17.868 | [1643973251.067850][65:75] CHIP:DMG: 					
CHIPTOOL | 2022-02-04 11:14:17.869 | [1643973251.067884][65:75] CHIP:DMG: 					Data = 25,
CHIPTOOL | 2022-02-04 11:14:17.983 | [1643973251.067913][65:75] CHIP:DMG: 			},
CHIPTOOL | 2022-02-04 11:14:17.985 | [1643973251.067950][65:75] CHIP:DMG: 			
CHIPTOOL | 2022-02-04 11:14:17.987 | [1643973251.067978][65:75] CHIP:DMG: 		},
CHIPTOOL | 2022-02-04 11:14:17.988 | [1643973251.068016][65:75] CHIP:DMG: 		
CHIPTOOL | 2022-02-04 11:14:17.989 | [1643973251.068044][65:75] CHIP:DMG: 	],
CHIPTOOL | 2022-02-04 11:14:17.990 | [1643973251.068084][65:75] CHIP:DMG:
CHIPTOOL | 2022-02-04 11:14:18.093 | [1643973251.068114][65:75] CHIP:DMG: 	SuppressResponse = true, 
CHIPTOOL | 2022-02-04 11:14:18.095 | [1643973251.068140][65:75] CHIP:DMG: }
CHIPTOOL | 2022-02-04 11:14:18.211 | [1643973251.068362][65:75] CHIP:TOO:  ***** Test Failure: Expecting success response but got a failure response
DEBUG    | 2022-02-04 11:14:18.213 | Found failure:  Expecting success response but got a failure response
WARNING  | 2022-02-04 11:14:18.214 | Test Failure:  Expecting success response but got a failure response
CHIPTOOL | 2022-02-04 11:14:18.215 | 
CHIPTOOL | 2022-02-04 11:14:18.217 | [1643973251.068476][65:75] CHIP:EM: Sending Standalone Ack for MessageCounter:13013002 on exchange 35898i
[... cut ...]
CHIPTOOL | 2022-02-04 11:14:18.532 | [1643973251.069109][65:65] CHIP:-: ../../commands/tests/TestCommand.cpp:85: CHIP Error 0x000000AC: Internal error at ../../commands/common/CHIPCommand.cpp:59
CHIPTOOL | 2022-02-04 11:14:18.534 | [1643973251.069165][65:65] CHIP:TOO: Run command failure: ../../commands/tests/TestCommand.cpp:85: CHIP Error 0x000000AC: Internal error
CHIPTOOL | 2022-02-04 11:14:18.638 | [1643973251.074355][65:65] CHIP:SPT: VerifyOrDie failure at ../../third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
INFO     | 2022-02-04 11:14:19.035 | Test Step Completed [FAILED]: read the mandatory attribute: MeasuredValue
```

woody-apple, 2022-02-08:
@AntonGrey Are there automated tests for this?

kowsisoundhar12, 2022-02-09:
@AntonGrey I have ran testscripts in CI ,Above mentioned testscripts are  working fine in CI ,please refer Attached log here
[TestExecution-status.zip](https://github.com/project-chip/connectedhomeip/files/8030264/TestExecution-status.zip)

AntonGrey, 2022-02-09:
@woody-apple Thanks. There are automated tests for this. Could you look below for what I found? 
@kowsisoundhar12 Thanks. But I see warning on checking measuredValue attribute `Warning: measuredValue type checking is not implemented yet. Expected type: 'int16'`.
So it looks like we still can't read measuredValue attr yet. Am I right?

Part of CI log.
```
2022-02-09 12:23:41.856 DEBUG   [1644389621.808913][51868:51872] CHIP:DMG: 	AttributeReportIBs =
2022-02-09 12:23:41.856 DEBUG   [1644389621.751704][51854:51854] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
2022-02-09 12:23:41.856 DEBUG   [1644389621.745052][51861:51865] CHIP:DMG: 				CommandPathIB =
2022-02-09 12:23:41.856 DEBUG   [1644389621.808919][51868:51872] CHIP:DMG: 	[
2022-02-09 12:23:41.856 DEBUG   [1644389621.751713][51854:51854] CHIP:IN: 0x04, 0x72, 0x98, 0x18, 0x70, 0xb5, 0xa6, 0x13, 
2022-02-09 12:23:41.856 DEBUG   [1644389621.745055][51861:51865] CHIP:DMG: 				{
2022-02-09 12:23:41.856 DEBUG   [1644389621.808922][51868:51872] CHIP:DMG: 		AttributeReportIB =
2022-02-09 12:23:41.856 DEBUG   [1644389621.751716][51854:51854] CHIP:IN: 0x82, 0xde, 0x0a, 0x56, 0xe9, 0x2e, 0xa8, 0xf0, 
2022-02-09 12:23:41.856 DEBUG   [1644389621.745058][51861:51865] CHIP:DMG: 					EndpointId = 0x0,
2022-02-09 12:23:41.857 DEBUG   [1644389621.808928][51868:51872] CHIP:DMG: 		{
2022-02-09 12:23:41.857 DEBUG   [1644389621.751719][51854:51854] CHIP:IN: 0x71, 0x4b, 0x38, 0xce, 0x1b, 0xdb, 0x30, 0x1b, 
2022-02-09 12:23:41.857 DEBUG   [1644389621.745061][51861:51865] CHIP:DMG: 					ClusterId = 0x3e,
2022-02-09 12:23:41.857 DEBUG   [1644389621.808931][51868:51872] CHIP:DMG: 			AttributeDataIB =
2022-02-09 12:23:41.857 DEBUG   [1644389621.751723][51854:51854] CHIP:IN: 0x8f, 0xd2, 0xc0, 0xcb, 0x3a, 0xd5, 0x4b, 0x0c, 
2022-02-09 12:23:41.857 DEBUG   [1644389621.745064][51861:51865] CHIP:DMG: 					CommandId = 0x1,
2022-02-09 12:23:41.857 DEBUG   [1644389621.808935][51868:51872] CHIP:DMG: 			{
2022-02-09 12:23:41.857 DEBUG   [1644389621.751727][51854:51854] CHIP:IN: 0xd5, 0xf6, 0x93, 0xb6, 0x21, 0xd8, 0x2b, 0x9b, 
2022-02-09 12:23:41.857 DEBUG   [1644389621.745067][51861:51865] CHIP:DMG: 				},
2022-02-09 12:23:41.857 DEBUG   [1644389621.808939][51868:51872] CHIP:DMG: 				DataVersion = 0xc1400626,
2022-02-09 12:23:41.857 DEBUG   [1644389621.751732][51854:51854] CHIP:IN: 0x74, 0x02, 0x1a, 0x13, 0x61, 0x32, 0xd8, 0xab, 
2022-02-09 12:23:41.857 DEBUG   [1644389621.745072][51861:51865] CHIP:DMG: 				
2022-02-09 12:23:41.857 DEBUG   [1644389621.808942][51868:51872] CHIP:DMG: 				AttributePathIB =
2022-02-09 12:23:41.857 DEBUG   [1644389621.751736][51854:51854] CHIP:IN: 0x1e, 0x25, 0xfe, 0x25, 0x7c, 0x4d, 0xa6, 0xa3, 
2022-02-09 12:23:41.857 DEBUG   [1644389621.745076][51861:51865] CHIP:DMG: 				CommandData = 
2022-02-09 12:23:41.858 DEBUG   [1644389621.808946][51868:51872] CHIP:DMG: 				{
2022-02-09 12:23:41.858 DEBUG   [1644389621.751740][51854:51854] CHIP:IN: 0xd4, 0xc6, 0xf1, 0xbb, 0x07, 0xbc, 0x95, 0xc9, 
2022-02-09 12:23:41.858 DEBUG   [1644389621.745080][51861:51865] CHIP:DMG: 				{
2022-02-09 12:23:41.858 DEBUG   [1644389621.808950][51868:51872] CHIP:DMG: 					Endpoint = 0x1,
2022-02-09 12:23:41.858 DEBUG   [1644389621.751743][51854:51854] CHIP:IN: 0x0d, 
2022-02-09 12:23:41.858 DEBUG   [1644389621.745085][51861:51865] CHIP:DMG: 					0x0 = [
2022-02-09 12:23:41.858 DEBUG   [1644389621.808954][51868:51872] CHIP:DMG: 					Cluster = 0x402,
2022-02-09 12:23:41.858 DEBUG   [1644389621.751761][51854:51854] CHIP:IN: Generated compressed fabric ID
2022-02-09 12:23:41.858 DEBUG   [1644389621.745088][51861:51865] CHIP:DMG: 						... (byte string too long) ...
2022-02-09 12:23:41.858 DEBUG   [1644389621.808958][51868:51872] CHIP:DMG: 					Attribute = 0x0000_0000,
2022-02-09 12:23:41.858 DEBUG   [1644389621.751764][51854:51854] CHIP:IN: 0x9f, 0x4d, 0x3a, 0x02, 0x6d, 0xb1, 0xe1, 0xcd, 
2022-02-09 12:23:41.858 DEBUG   [1644389621.745091][51861:51865] CHIP:DMG: 				]
2022-02-09 12:23:41.858 DEBUG   [1644389621.808962][51868:51872] CHIP:DMG: 				}
2022-02-09 12:23:41.858 DEBUG   [1644389621.751770][51854:51854] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
2022-02-09 12:23:41.859 DEBUG   [1644389621.745095][51861:51865] CHIP:DMG: 					0x1 = [
2022-02-09 12:23:41.859 DEBUG   [1644389621.808966][51868:51872] CHIP:DMG: 					
2022-02-09 12:23:41.859 DEBUG   [1644389621.751773][51854:51854] CHIP:DIS: Assigned compressed fabric ID: 0x9F4D3A026DB1E1CD, node ID: 0x0000000012344321
2022-02-09 12:23:41.859 DEBUG   [1644389621.745106][51861:51865] CHIP:DMG: 						0xde, 0x7f, 0x5d, 0x64, 0xc2, 0x6e, 0x25, 0xc0, 0x6c, 0xb0, 0xd3, 0x16, 0x5d, 0x4d, 0xc8, 0xb3, 0xe4, 0x22, 0x32, 0xa3, 0x7e, 0x6b, 0xb8, 0x25, 0x8f, 0x7, 0xf4, 0x6a, 0x9, 0x86, 0x71, 0x83, 0x8a, 0x50, 0xb1, 0x37, 0x35, 0x9a, 0x2b, 0x23, 0xaa, 0x17,
2022-02-09 12:23:41.859 DEBUG   [1644389621.808972][51868:51872] CHIP:DMG: 					Data = -32768, 
2022-02-09 12:23:41.859 DEBUG   [1644389621.751776][51854:51854] CHIP:DIS: Fabric (1) loaded from storage. Calling OnFabricRetrievedFromStorage
2022-02-09 12:23:41.859 DEBUG   [1644389621.745110][51861:51865] CHIP:DMG: 				]
2022-02-09 12:23:41.859 DEBUG   [1644389621.808975][51868:51872] CHIP:DMG: 			},
2022-02-09 12:23:41.859 DEBUG   [1644389621.751779][51854:51854] CHIP:ZCL: OpCreds: Fabric 0x1 was retrieved from storage. FabricId 0x0000000000000001, NodeId 0x0000000012344321, VendorId 0xFFF1
[... cut ...]
2022-02-09 12:23:41.861 DEBUG   [1644389621.745142][51861:51865] CHIP:DMG: 	
2022-02-09 12:23:41.861 DEBUG   [1644389621.809009][51868:51872] CHIP:DMG: }
2022-02-09 12:23:41.861 DEBUG   [1644389621.752170][51854:51854] CHIP:ZCL: OpCreds: Call to fabricListChanged
2022-02-09 12:23:41.861 DEBUG   [1644389621.745146][51861:51865] CHIP:DMG: 	InteractionModelRevision = 1
2022-02-09 12:23:41.861 DEBUG   [1644389621.809045][51868:51872] CHIP:TOO: Warning: measuredValue type checking is not implemented yet. Expected type: 'int16'
```

raju-apple, 2022-02-09:
@AntonGrey That warning means that we still cannot validate that the measuredValue 'type' can't be validated yet. For reading the actual value , we are reading the value but we aren't checking that value against a specific value because there isn't a default value for 'measured value' defined in the spec. What we could do is write a value and try to read the same value again , if that is the way forward we could request the test plan team to make that change and we can update the scripts as well. Thoughts ?

AntonGrey, 2022-02-09:
@raju-apple Thanks for explanation. I believe we can't write measuredValue because it is readonly attribute according to specification. No additional checks are required. 

So it looks like original issue is fixed at master branch and ticket can be closed.

closed 2022-02-09 by raju-apple