project-chip/connectedhomeip #14823: [TE7.5] [TC-MF-1.11] [TC-MF-1.12] It happened to crash on Controlee using an invalid passcode

wissn84, 2022-02-07:
#### Problem
- DUT_CR1: our-controller
- TH_CR2: chip-tool
- TH_CE: all-clusters-app

In test case, 
Set up a DUT_CR1 to start attempting to do PASE to TH_CE and failing 20 times. This can be done using an invalid passcode.
If using an invalid passcode, TH_CE happened to crash.

TH Log 
`[1644200531.326835][4946:4946] CHIP:EM: Received message of type 0x20 with protocolId (0, 0) and MessageCounter:3319376779 on exchange 48621r
[1644200531.326949][4946:4946] CHIP:EM: Handling via exchange: 48621r, Delegate: 0xaaaac56651e0
[1644200531.327050][4946:4946] CHIP:SC: Received PBKDF param request
[1644200531.327144][4946:4946] CHIP:SC: Peer assigned session ID 1
[1644200531.327211][4946:4946] CHIP:SC: Found MRP parameters in the message
[1644200531.327376][4946:4946] CHIP:SC: Including MRP parameters in PBKDF param response
[1644200531.329093][4946:4946] CHIP:EM: Piggybacking Ack for MessageCounter:3319376779 on exchange: 48621r
[1644200531.329187][4946:4946] CHIP:IN: Prepared plaintext message 0xaaaac5664370 to 0x0000000000000000 (0)  of type 0x21 and protocolId (0, 0) on exchange 48621r with MessageCounter:2358684095.
[1644200531.329274][4946:4946] CHIP:IN: Sending plaintext msg 0xaaaac5664370 with MessageCounter:2358684095 to 0x0000000000000000 at monotonic time: 2334234 msec
[1644200531.329516][4946:4946] CHIP:SC: Sent PBKDF param response
[1644200531.410951][4946:4946] CHIP:EM: Received message of type 0x22 with protocolId (0, 0) and MessageCounter:3319376780 on exchange 48621r
[1644200531.411045][4946:4946] CHIP:EM: Found matching exchange: 48621r, Delegate: 0xaaaac56651e0
[1644200531.411141][4946:4946] CHIP:EM: Rxd Ack; Removing MessageCounter:2358684095 from Retrans Table on exchange 48621r
[1644200531.411201][4946:4946] CHIP:EM: Removed CHIP MessageCounter:2358684095 from RetransTable on exchange 48621r
[1644200531.411277][4946:4946] CHIP:SC: Received spake2p msg1
[1644200531.414970][4946:4946] CHIP:EM: Piggybacking Ack for MessageCounter:3319376780 on exchange: 48621r
[1644200531.415058][4946:4946] CHIP:IN: Prepared plaintext message 0xaaaac5664370 to 0x0000000000000000 (0)  of type 0x23 and protocolId (0, 0) on exchange 48621r with MessageCounter:2358684096.
[1644200531.415142][4946:4946] CHIP:IN: Sending plaintext msg 0xaaaac5664370 with MessageCounter:2358684096 to 0x0000000000000000 at monotonic time: 2334320 msec
[1644200531.415349][4946:4946] CHIP:SC: Sent spake2p msg2
[1644200531.553467][4946:4946] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:3319376781 on exchange 48621r
[1644200531.553542][4946:4946] CHIP:EM: Found matching exchange: 48621r, Delegate: 0xaaaac56651e0
[1644200531.553614][4946:4946] CHIP:EM: Rxd Ack; Removing MessageCounter:2358684096 from Retrans Table on exchange 48621r
[1644200531.553654][4946:4946] CHIP:EM: Removed CHIP MessageCounter:2358684096 from RetransTable on exchange 48621r
[1644200531.553720][4946:4946] CHIP:SC: Received error (protocol code 2) during PASE process. ../../src/protocols/secure_channel/PASESession.cpp:858: CHIP Error 0x00000038: Invalid PASE parameter
[1644200531.553795][4946:4946] CHIP:SC: Failed during PASE session setup. ../../src/protocols/secure_channel/PASESession.cpp:858: CHIP Error 0x00000038: Invalid PASE parameter
[1644200531.553842][4946:4946] CHIP:SVR: Commissioning failed (attempt 1): ../../src/protocols/secure_channel/PASESession.cpp:858: CHIP Error 0x00000038: Invalid PASE parameter`

 #### Proposed Solution
TH don't happen to crash

#### Full Log
[TE7_5-TC-MF-1.12_controllee_crashed.txt](https://github.com/project-chip/connectedhomeip/files/8012397/TE7_5-TC-MF-1.12_controllee_crashed.txt)

[TE7_5-TC-MF-1.11_controllee_crashed.txt](https://github.com/project-chip/connectedhomeip/files/8012396/TE7_5-TC-MF-1.11_controllee_crashed.txt)

andy31415, 2022-05-30:
@jeonghwan7 - could you validate this in ToT?

TE7.5 is quite old.

andy31415, 2022-11-03:
Can we close this? This was reported against TC-MF which was renamed to TC-CADMIN in #20514. We still run Test_TC_CADMIN_1_11.yaml and Test_TC_CADMIN_1_12.yaml during test events past 7.5.

andy31415, 2022-11-03:
Also title sais "crash" however logs seem to say "invalid parameter". Is this really a crash?

closed 2023-04-18 by jeonghwan7