project-chip/connectedhomeip #14825: [TC-I-1.1]- In Identify Cluster, the cluster revision value is not matched with the expected outcome.

Rajashreekalmane, 2022-02-07:
In Identify Cluster, testcase 1.1 (TC-I-1.1), the "cluster revision attribute" value must be 4 as per the Test Plan,
But the actual outcome is 2. This needs to be fixed as per Test plan expected outcome.

expected behavior - cluster revision attribute - 4
actual behavior - cluster revision - 2

Test Test-plan link:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/identify.adoc

Actual command:- ./chip-tool identify read cluster-revision 1 1
Actual log is attached below
[[TC-I-1.1].txt](https://github.com/project-chip/connectedhomeip/files/8012768/TC-I-1.1.txt)

Survensa, 2022-03-22:
Same issue is reflected in thread connectivity.

Survensa, 2022-04-04:
The issue is resolved in latest commit 41a431dcc7c123699818636afd01b57412a6f3a4

closed 2022-04-04 by Rajashreekalmane