project-chip/connectedhomeip #14833: Bootstrap on Apple Silicon / ARM compatible version of protoc missing

dtodor, 2022-02-07:
The bootstrap script fails on Apple Silicon:
```
➜  connectedhomeip.git git:(master) ✗ source scripts/activate.sh

  WELCOME TO...

           ░▓░
           ▓█▓
           ▓█▓                                                     ▒█     ▒█
      ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
       ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
    ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
    ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
     ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
 ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓
 ░██▒░  ▒██▒ ▒██░  ░▒█▓
         ░▓░ ░▓░

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (2.4s)
  Setting up Python environment.....[/]
ninja: Entering directory `/Users/anasoft/Downloads/connectedhomeip.git/.environment/gn_out'
[1/207] touch python/obj/third_party/pigweed/repo/pw_presubmit/py/py._run_pip_install.stamp
[2/207] touch python/obj/third_party/pigweed/repo/pw_presubmit/py/py.install.stamp
[3/207] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_tokenizer --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_protobuf_compiler/toolchain:protocol_buffer --capture-output -- ../../third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py --language python --include-file protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/includes.txt --compile-dir protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/sources --out-dir protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/python --sources protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/sources/pw_tokenizer/proto/options.proto
FAILED: protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/python/pw_tokenizer/proto/options_pb2.py protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/python/pw_tokenizer/proto/options_pb2.pyi 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_tokenizer --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_protobuf_compiler/toolchain:protocol_buffer --capture-output -- ../../third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py --language python --include-file protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/includes.txt --compile-dir protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/sources --out-dir protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/python --sources protocol_buffer/gen/third_party/pigweed/repo/pw_tokenizer/proto.proto_library/sources/pw_tokenizer/proto/options.proto
Traceback (most recent call last):
  File "../../third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py", line 201, in <module>
    sys.exit(main())
  File "../../third_party/pigweed/repo/pw_protobuf_compiler/py/pw_protobuf_compiler/generate_protos.py", line 183, in main
    process = subprocess.run(cmd,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 86] Bad CPU type in executable: 'protoc'
```

dtodor, 2022-02-07:
A workaround is to install the Protocol Buffer Compiler v3.17.3 manually. I've tried this older brew formula, which seems to do the job:

[https://github.com/Homebrew/homebrew-core/blob/ac2a2846981f6c6f9d6df64b9250a1e9f872cbf2/Formula/protobuf.rb](url)

bzbarsky-apple, 2022-05-21:
@dtodor Is this still an issue after recent pigweed updates?

dtodor, 2022-05-21:
I removed the homebrew version of protoc and did a clean bootstrap - everything seems to work now:
```
➜  ~ which protoc
/Users/anasoft/Downloads/connectedhomeip.git/.environment/cipd/packages/pigweed/bin/protoc
```

dtodor, 2022-05-21:
Ooops, sorry, it is still the x86_64 version:

```
➜  ~ file /Users/anasoft/Downloads/connectedhomeip.git/.environment/cipd/packages/pigweed/bin/protoc
/Users/anasoft/Downloads/connectedhomeip.git/.environment/cipd/packages/pigweed/bin/protoc: Mach-O 64-bit executable x86_64
```
It works because I have Rosetta installed.

bzbarsky-apple, 2022-05-23:
@andy31415 Does pigweed support this?

andy31415, 2022-05-23:
This may have been fixed since then.

woody-apple, 2022-05-24:
Works for me.

closed 2022-05-24 by woody-apple

mustafayuce33, 2023-09-23:
@woody-apple @bzbarsky-apple  This issue still happens on my macbook pro M2 Max. As @dtodor said it still uses x86_64. Are you sure that this has been resolved?