project-chip/connectedhomeip #14837: ESP32 bridge app example doesn't compile with esp-idf v4.4: conflicting definitions for htons, ntons, htonl, ntonl

SgtPepperFTW, 2022-02-07:
#### Problem
when compiling esp32 example the definitions of htons, ntons, htonl, ntonl are conflicted from esp-idff component in "components/lwip/port/esp32/include/arch/cc.h"
 and matter file "src/inet/arpa-inet-compatibility.h"

 #### Proposed Solution
put another define in arpa-inet-compatibility? remove these defines from arpa-inet-compatibility alltogether?

dhrishi, 2022-02-07:
Hi @SgtPepperFTW I just tried compilation of `examples/bridge-app/esp32` on recent CHIP commit `0f7be8bef3a6d1df72c3a2f6ea7c0c359fd44937` with IDF v4.4 release (`8153bfe4125e6a608abccf1561fd10285016c90a`). I am able to successfully compile it. Can you please clean the app and retry?

SgtPepperFTW, 2022-02-07:
@dhrishi yeah, my bad, I was on the master branch of the esp-idf, which is apparently v5.0-dev. Still means it might be an issue later.

dhrishi, 2022-02-07:
>Still means it might be an issue later.

Yes. Looks like we will have to handle it with v5.0 or later. I will close this for now.

closed 2022-02-07 by dhrishi