project-chip/connectedhomeip #14840: [TC-GENDIAG-1.1]IPv4 and IPv6 address  missing in the outcome when we read network-interfaces attribute in General Diagnostics cluster.

Rajashreekalmane, 2022-02-07:
#### Problem
In General Diagnostics Cluster when we pass "./chip-tool generaldiagnostics read network-interfaces 1 0" command the expected outcome is "The read data type must match the value listed in the specification(s) Table 118. NetworkInterface ENUM.
-  Structure data includes Name (interface name)
- FabricConnected 
- OffPremiseServicesReaschableIPv4 
- OffPremiseServicesReaschableIPv6 
-  HardwareAddress
- IPv4Address
- IPv6Address 
- Type 

But IPv4Address and IPv6Address are not present in result 

PFA actual log
[[TC-GENDIAG-1.1].txt](https://github.com/project-chip/connectedhomeip/files/8016095/TC-GENDIAG-1.1.txt)

Rajashreekalmane, 2022-02-16:
According to the spec, we should get   IPV4 and IPv6 addresses as list  addresses that are currently assigned to the network interface, but we am getting the response as 0 entries.
PFA screenshot:
[image]

Survensa, 2022-03-22:
Same issue is reflected in thread connectivity for IPV6.