project-chip/connectedhomeip #14848: chip-tool fails to initiate AnnounceOTAProvider command when subscribed to events on OTA-Requestor Linux app

ukatira, 2022-02-07:
#### Problem
1. Launch and Commission OTA-Provider and OTA-Requestor reference Linux apps
2. Subscribe events on OTA-R Linux app using chip-tool
3. Initiate query using AnnounceOTAProvider command from chip-tool

chip-tool fails to initiate AnnounceOTAProvider command with error:
[1644258223657] [3613:3389969] CHIP: [-] ../../../examples/chip-tool/commands/common/CHIPCommand.cpp:248: CHIP Error 0x00000032: Timeout at ../../../examples/chip-tool/commands/common/CHIPCommand.cpp:61
[1644258223657] [3613:3389969] CHIP: [TOO] Run command failure: ../../../examples/chip-tool/commands/common/CHIPCommand.cpp:248: CHIP Error 0x00000032: Timeout

 #### Proposed Solution
This works when AnnounceOTAProvider command is triggered before subscribing events but should work the other way as well so events can be subscribed and received right from the beginning of OTA process.

[chiptool_announce_ota.txt](https://github.com/project-chip/connectedhomeip/files/8017842/chiptool_announce_ota.txt)
[chiptool_event_subscribe.txt](https://github.com/project-chip/connectedhomeip/files/8017843/chiptool_event_subscribe.txt)
[ota-p.txt](https://github.com/project-chip/connectedhomeip/files/8017845/ota-p.txt)
[ota-r.txt](https://github.com/project-chip/connectedhomeip/files/8017846/ota-r.txt)

bzbarsky-apple, 2022-02-07:
What exact commands are being run?  Are the two chip-tool instances running at the same time, on the same IP address, using the same fabric?  If so, they are both going to try to use the same port, and that's not going to work very well....

ukatira, 2022-02-07:
Command for AnnounceOTAProvider command:
chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 0 2 0
Command to subscribe events on OTA-Requestor app:
chip-tool otasoftwareupdaterequestor subscribe-event state-transition 5 10 1 2 0

Yes the two instances are running at the same time, on the same IP address, using the same fabric.
@bzbarsky-apple Do you know a workaround to try out to make this work in the required sequence?

bzbarsky-apple, 2022-02-07:
There's no workaround without code changes right now.

@vivien-apple maybe we should add a way to run chip-tool with a non-default (or even ephemeral) port?  Might still break, because the same node running on multiple ports is not really something that's OK per spec.....

The other option would be to have some sort of shell/repl thing in chip-tool so you can issue it commands while it's listening for reports....

carol-apple, 2022-02-07:
@bzbarsky-apple I saw mentions of running two Linux all-cluster-app in different docker containers in https://github.com/project-chip/connectedhomeip/pull/14303

Running multiple Linux apps in separate IP namespaces should solve the issue? If so, could you provide instructions on how to do that here?

bzbarsky-apple, 2022-02-08:
You'd need the two apps to be on the same fabric (as in, using the same certificate store, etc).  I guess as long as they are not both trying to write to it that might work...

IP namespaces might do the trick, but all I don't really know anything about the specifics of how to set it up....

isiu-apple, 2022-04-27:
I ran the sequence of command as reported in this issue but I do not observe the reported timeout issue.  I see that Query Image succeeds and event report generation continues to run even if I am running 2 instances of chiptool. 

1. Launch and Commission OTA-Provider and OTA-Requestor reference Linux apps
2. Subscribe events on OTA-R Linux app using chip-tool
3. Initiate query using AnnounceOTAProvider command from chip-tool

Terminal 1
`out/apps/ota-provider/chip-ota-provider-app -f ~/Downloads/test2.ota --passcode 20202021 --discriminator 3840 --KVS /tmp/chip_kvs_provider`

Terminal 2
`out/apps/ota-requestor/chip-ota-requestor-app --discriminator 30 --secured-device-port 5560 --KVS /tmp/chip_kvs_requestor
`

Terminal 3
`out/apps/chip-tool/chip-tool otasoftwareupdaterequestor subscribe-event state-transition 5 10 0x1234567890 0 --timeout 65535`

Terminal 4
`out/apps/chip-tool/chip-tool otasoftwareupdaterequestor announce-ota-provider 0x00000000ABCD 0 0 0 0x0000001234567890 0`

bzbarsky-apple, 2022-04-28:
Yes, after https://github.com/project-chip/connectedhomeip/pull/16452 chip-tool uses ephemeral ports, so this works now.

closed 2022-04-28 by bzbarsky-apple

isiu-apple, 2022-04-28:
This PR (https://github.com/project-chip/connectedhomeip/pull/16452) changed chiptool to use ephemeral ports instead of static ports so that multiple instances of chiptool can be run using the same fabric in parallel on the same laptop/IP address.  Therefore the issue reported in this ticket has now been resolved.