project-chip/connectedhomeip #14859: Crash in chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets, 32u>::~EndPointManagerImplPool

woody-apple, 2022-02-07:
#### Problem
```
Thread 6 name:   Dispatch queue: com.apple.CoreHAP.CHIPAccessoryServer.clientQueue.4445458304
Thread 6 Crashed:
0   libsystem_kernel.dylib        	       0x1c1ed8b98 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x1fcb5f3dc pthread_kill + 272
2   libsystem_c.dylib             	       0x195246538 abort + 168
3   CHIP                          	       0x110e1ba08 chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets, 32u>::~EndPointManagerImplPool() + 164
4   CHIP                          	       0x110e1a208 chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets, 32u>::~EndPointManagerImplPool() + 32
5   libsystem_c.dylib             	       0x195246e9c __cxa_finalize_ranges + 464
6   libsystem_c.dylib             	       0x195241718 exit + 32
7   XXXX                    	       0x1017192f8 -[XXXX XXXX:] + 532
8   XXXX                  	       0x101208798 -[XXXXX] + 148
9   XXXX                  	       0x1012072e4 ____activateIfNecessary_block_invoke + 40
10  libdispatch.dylib             	       0x18990b8dc _dispatch_block_async_invoke2 + 148
11  libdispatch.dylib             	       0x1898fcad8 _dispatch_client_callout + 20
12  libdispatch.dylib             	       0x1898fff98 _dispatch_continuation_pop + 500
13  libdispatch.dylib             	       0x18991313c _dispatch_source_invoke + 1596
14  libdispatch.dylib             	       0x1899040ac _dispatch_lane_serial_drain + 376
15  libdispatch.dylib             	       0x189904d60 _dispatch_lane_invoke + 444
16  libdispatch.dylib             	       0x18990f5ac _dispatch_workloop_worker_thread + 648
17  libsystem_pthread.dylib       	       0x1fcb59140 _pthread_wqthread + 288
18  libsystem_pthread.dylib       	       0x1fcb58ee0 start_wqthread + 8
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-06-13:
@woody-apple Are we still seeing this?

bzbarsky-apple, 2022-10-12:
This is still happening.  Something is leaking `UDPEndPointImplSockets` and then we crash on shutdown....

bzbarsky-apple, 2023-01-31:
Still happening, with stacks like:
```
3   Matter                        	       0x1b62f6ff0 chipAbort + 20 (/Library/Caches/com.apple.xbs/Binaries/CHIPFramework/install/TempContent/Objects/Matter.build/Matter.build/out/../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/lib/support/CodeUtils.h:508)
4   Matter                        	       0x1b64d56bc chip::Inet::EndPointManager<chip::Inet::UDPEndPoint>::~EndPointManager() + 376 (/Library/Caches/com.apple.xbs/Binaries/CHIPFramework/install/TempContent/Objects/Matter.build/Matter.build/out/../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/inet/InetLayer.h:59)
5   Matter                        	       0x1b64d5534 chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets>::~EndPointManagerImplPool() + 76 (/Library/Caches/com.apple.xbs/Binaries/CHIPFramework/install/TempContent/Objects/Matter.build/Matter.build/out/../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/inet/InetLayer.h:119)
6   Matter                        	       0x1b64d4834 chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets>::~EndPointManagerImplPool() + 32 (/Library/Caches/com.apple.xbs/Binaries/CHIPFramework/install/TempContent/Objects/Matter.build/Matter.build/out/../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/inet/InetLayer.h:119)
7   libsystem_c.dylib             	       0x1858f4a6c __cxa_finalize_ranges + 476 (/Library/Caches/com.apple.xbs/Sources/Libc/stdlib/FreeBSD/atexit.c:290)
```
due to `exit` calls.

closed 2023-02-01 by woody-apple