project-chip/connectedhomeip #14865: SoftwareVersion CLI option for OTA-P Linux app does not accept X.Y format

ukatira, 2022-02-08:
#### Problem
When using SoftwareVersion CLI option for OTA-P Linux app to set any X.Y value, it only picks up and sends X in QueryImage Response
Command: chip-ota-provider-app -f test.bin -s 3.5
OTA-R trace shows:
[1644283500022] [16877:3760320] CHIP: [SWU] QueryImageResponse:
[1644283500022] [16877:3760320] CHIP: [SWU]   status: 0
[1644283500022] [16877:3760320] CHIP: [SWU]   delayedActionTime: 0 seconds
[1644283500022] [16877:3760320] CHIP: [SWU]   imageURI: bdx://0000000000000001/test.bin
[1644283500022] [16877:3760320] CHIP: [SWU]   softwareVersion: 3
[1644283500022] [16877:3760320] CHIP: [SWU]   softwareVersionString: Example-Image-V0.1
[1644283500022] [16877:3760320] CHIP: [SWU]   updateToken: 32
[1644283500022] [16877:3760320] CHIP: [SWU]   userConsentNeeded: 0
[1644283500022] [16877:3760320] CHIP: [SWU]   metadataForRequestor: 0

 #### Proposed Solution
SoftwareVersion CLI option for OTA-P Linux app should accept any format supported by Matter

shubhamdp, 2022-02-08:
> Ref: 11.20.6.8. QueryImageResponse Command

SoftwareVersion is of type uint32, that means only integer values are valid.

If we want something like "X.Y", then I think that should go in SoftwareVersionString.

carol-apple, 2022-02-08:
@shubhamdp I'm thinking one of two options here. We should either:
1. Use the `-s` option for both `softwareVersion` and `softwareVersionString` (maybe just stringify 3 to "3.0")
2. Pass in an additional command line option for the SoftwareVersionString