project-chip/connectedhomeip #14886: [ESP32 bridge-app] How to start http server in parallel to matter stack?

VOMediola, 2022-02-08:
When including "esp_http_server.h" in main I get no file found. Compiling command doesn't include all of esp-idf components, where can I change the build parameters to include http_server component?

dhrishi, 2022-02-08:
@VOMediola Can you please add `esp_http_server` entry [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/bridge-app/esp32/main/CMakeLists.txt#L50) in the bridge-app?

VOMediola, 2022-02-09:
@dhrishi  thank you, that works. Can you exaplain why I need to require that, but not, for example, nvs_flash or esp_netif as a component?

Also, can you tell me what should I pay attention to when trying to use http server with matter? Do I need to init netif and create default event loop, or is it handled somwhere in Matter stack?

closed 2022-09-22 by stale[bot]