project-chip/connectedhomeip #14888: Spec provisions about errors for attempts to read/subscribe unsupported events are not implemented

bzbarsky-apple, 2022-02-08:
#### Problem
The spec says, in "8.4.3.2. Incoming Read Request and Subscribe Request Action Processing":

> Each request path in the EventRequests field SHALL be processed as follows:
>
>    If the path is a concrete path:
>
>       If the path indicates a cluster event that is not supported, an EventStatusIB SHALL be generated with the UNSUPPORTED_EVENT Status Code.

but the SDK has no such thing.  I see nothing in `Engine::BuildSingleReportDataEventReports` that checks for support, there is no way to list events as supported or not supported, and both `EMBER_ZCL_STATUS_UNSUPPORTED_EVENT` and `Status::UnsupportedEvent` are unused.

 #### Proposed Solution
Implement what the spec says.

@yunhanw-google @mrjerryjohns

mrjerryjohns, 2022-02-10:
I thought we'd discussed this ad nauseum quite a number of times. We're not actually implementing support in the SDK to generate the list of 'supported events' for a given target (which would be a pre-req for generating such an error) for v1.0.

bzbarsky-apple, 2022-02-10:
We discussed whether we are generating the EventList attribute.  That's not the same as silently pretending like there is a subscription to an event that will not in fact ever happen....

mrjerryjohns, 2022-03-03:
The pre-requisites for both generating the EventList attribute and rejecting a path in a subscription to an unsupported events are the same - you need knowledge of the set of implemented events on a given target, which is what I thought we had agreed we won't implement for v1.0.

bzbarsky-apple, 2023-06-26:
Partially fixed in #26761.