project-chip/connectedhomeip #14897: Support various list write operations in chip-tool

carol-apple, 2022-02-08:
#### Problem
Currently, there is no way to specify the various list write operations (section 8.9.2.7 ListDataWriteIB) via chip-tool

 #### Proposed Solution
@vivien-apple suggested maybe dedicated commands like:
- `list-replace`
- `list-delete`
- `list-modify`
- `list-add`

For `list-replace` and `list-add`, there should be no index param needed
For `list-modify` and `list-delete`, the indices param is needed
For `list-replace`, `list-modify`, and `list-add`, the entries (list contents) param is needed

bzbarsky-apple, 2022-02-08:
Note that there is no IM-level support, and none planned for 1.0, for list-delete, list-modify, and list-add.

The only thing supported is list-replace.

carol-apple, 2022-02-11:
Thanks @bzbarsky-apple. I'll move this out of V1.0