project-chip/connectedhomeip #14903: TPLink Archer C5400 router doesn't work well with Matter

mrjerryjohns, 2022-02-08:
#### Problem

When commissioning an Rpi with IPv6 enabled, it doesn't complete commissioning successfully. Specifically, the REPL acting as a commissioner is unable to connect operationally over CASE to the Rpi:

```
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.DL[59137] DEBUG Mdns: OnGetAddrInfo hostname:EEAABADABAD0DDCA.local.
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.DIS[59137] INFO Node ID resolved for 0x0000000000000002
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.DIS[59137] INFO     Addr 0: [::e65f:1ff:fe82:f926]:5540
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.CTL[59137] INFO F0000000000000001: OperationalDiscoveryComplete for device ID N0000000000000002
…
…
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.IN[59137] INFO Prepared unauthenticated message 0x7f86138df868 to 0x0000000000000000 (0)  of type 0x30 and protocolId (0, 0) on exchange 61170i with MessageCounter:1597295096.
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.IN[59137] INFO Sending unauthenticated msg 0x7f86138df868 with MessageCounter:1597295096 to 0x0000000000000000 at monotonic time: 959560209 msec
2022-02-07 13:25:10 johnsj-macbookpro1.roam.corp.google.com chip.CTL[59137] INFO Device connection failed. Error ../../src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000041: No route to host
```

The key failure is the inability to get a route to the Rpi.

#### Cause

The Router attempts to do prefix delegation, but has insufficient bits granted by my ISP to assign a prefix. So it grants the device a completely bogus global IPv6 address: `::e65f:1ff:fe82:f926`

This is picked up commissioner during mDNS discovery and it attempts to send a message to that address, but clearly fails.

 #### Proposed Solution

While we were able to prevent the router from doing this by switching the mode on the router from "SLAAC + Stateless DHCP" to "DHCPv6", we can add a heuristic to de-prioritize such an address in the `IsIpLess` function.

mrjerryjohns, 2022-02-08:
FYI @andreilitvin @tcarmelveilleux

mrjerryjohns, 2022-02-08:
Many thanks to @tcarmelveilleux for helping me debug this.

mrjerryjohns, 2022-02-08:
So, we only seem to resolve to 1 IP address in the logs above.

However, calling `dns-sd` on my Mac seems to show multiple addresses that are returned:

```
(python_env) ➜  connectedhomeip git:(ipv6/zero-prefix-addrs) ✗ dns-sd -Gv4v6  EEAABADABAD0DDCA.local.
DATE: ---Tue 08 Feb 2022---
12:08:12.938  ...STARTING...
Timestamp     A/R    Flags if Hostname                               Address                                      TTL
12:08:12.940  Add 40000003  6 EEAABADABAD0DDCA.local.                0000:0000:0000:0000:0000:0000:0000:1000%<0>  120
12:08:12.940  Add 40000003  6 EEAABADABAD0DDCA.local.                FE80:0000:0000:0000:E65F:01FF:FE82:F926%en0  120
12:08:12.940  Add 40000002  6 EEAABADABAD0DDCA.local.                192.168.0.190                                120
```

cecille, 2022-02-25:
@mrjerryjohns  - Andrei put up a PR a few days ago to preferentially select addresses. Can you re-test and see if this PR has fixed this issue?
https://github.com/project-chip/connectedhomeip/pull/15200/files

mrjerryjohns, 2022-02-25:
Will try it out. However, it doesn't look like multiple entries were even returned to begin with in the logs above?

andy31415, 2022-02-25:
I will update the code to use the new resolver, however for now the final code is not in. However could you verify if it 'would work'? 

Use:

```
./scripts/build/build_examples.py --target linux-x64-address-resolve-tool build

./out/linux-x64-address-resolve-tool/address-resolve-tool node 0x000000000001E1B9 0x116362B9D227A22A
```

and subsitute your fabricid and nodeid in there, see if the result is good. Specifically you should have a line like:

```
CHIP:DIS: Resolve completed: UDP:[2607:fea8:1fdf:1001:ea68:e7ff:fe31:1684%enp58s0]:5540
CHIP:DIS:    Supports TCP:                  YES
CHIP:DIS:    MRP IDLE retransmit timeout:   5000 ms
CHIP:DIS:    MRP ACTIVE retransmit timeout: 300 ms
CHIP:DIS: Discovery does not require any more timeouts
```

which will tell you the final result

andy31415, 2022-02-25:
I also have a pending PR in #15580 to only attach an interface to link local addresses that is not yet in master, so interface may not be final .But address selection should be (it should not pick the bogus 0/64 prefix I hope)

andy31415, 2022-03-23:
@mrjerryjohns is this still an issue with the latest SDK? We have heavily updated address resolve code now.

closed 2022-12-21 by stale[bot]