project-chip/connectedhomeip #14948: [ESP32 bridge-app] Endpoint 0 not found in IncreaseClusterDataVersion!

VOMediola, 2022-02-09:
Latest build throws following errors on start

```
I (775) chip[DL]: Device[Light 1]: ONLINE
E (775) chip[DMG]: Endpoint 0, Cluster 0x0000_0039 not found in IncreaseClusterDataVersion!
I (775) chip[DL]: Device[Light 2]: ONLINE
E (775) chip[DMG]: Endpoint 0, Cluster 0x0000_0039 not found in IncreaseClusterDataVersion!
I (785) chip[DL]: Device[Light 3]: ONLINE
E (795) chip[DMG]: Endpoint 0, Cluster 0x0000_0039 not found in IncreaseClusterDataVersion!
I (805) chip[DL]: Device[Light 4]: ONLINE
E (805) chip[DMG]: Endpoint 0, Cluster 0x0000_0039 not found in IncreaseClusterDataVersion!
```
Guess it has something to do with https://github.com/project-chip/connectedhomeip/pull/14471 only how to add DataVersion to Endpoint 0, which is not dynamic and is a "Bridge" cluster itself?

PSONALl, 2022-02-16:
@VOMediola this issue is solved on the latest CHIP HEAD d671a523745079b1ea1778ee4885159868d9c002

dhrishi, 2022-02-16:
@VOMediola Can you please confirm and close this issue if it is working as expected.

closed 2022-05-21 by bzbarsky-apple