project-chip/connectedhomeip #14954: [TC-MC-3.1]KeyPad Input Cluster-Getting Invalid Argument error for key code 1000 in Step 9

Rajashreekalmane, 2022-02-09:
#### Problem
As per the expectation, for the Media Cluster TC-MC-3.1 step 9, the response should be value 1 and unsupported key, in chip-tool as the controller and chip-tv-app as the accessory, getting an invalid argument error response. 

Tested with commit ID:-cab1ff9cf129ee7d2b79c225a4682fca2f0bc1e8

PFA Screenshot
[image]

closed 2022-02-18 by woody-apple

lazarkov, 2022-02-19:
@Rajashreekalmane Yes, you can only use codes that are defined in CEC protocol. There's no code 256 or 0x100 (in HEX value), from the list here: https://android.googlesource.com/platform/frameworks/base.git/+/master/services/core/java/com/android/server/hdmi/HdmiCecKeycode.java

Rajashreekalmane, 2022-02-21:
@lazarkov In test-Plan they are mentioned "TH sends <Any Unsupported> Navigation Keycode to DUT" for this we should get a response as "unsupported key (value 1) status response" in test step 9.

Test-Plan reference:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mc-3-1-navigation-keycode-verification.

Spec reference :
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/media/KeypadInput.adoc#data-types

lazarkov, 2022-02-24:
@Rajashreekalmane Any of the CEC predefined codes can be unsupported. The problem for the code defined in test case by number 1000. It's not that it's not supported, it's not valid. 

With current SDK you can choose number 10 for test-case it's not supported.