project-chip/connectedhomeip #14980: Unable to pair M5 via multi admin

woody-apple, 2022-02-09:
#### Problem
Unable to pair M5 with 2nd chip tool on iOS

Steps:
1. Paired 1st chip tool on iOS with M5 board and pairing acc and toggle light works fine. 
2. Enabled pairing mode on 1st chip tool and we do see the orange LED come up on M5 board
3. Launched 2nd chip tool on iOS and tried to pair the same M5 board into second chip tool via QR code scan. 
4. Selected “Go” after scanning but it does not proceed further. 
5. Seeing below errors on M5 logs. 

7140r
E (177577) chip[SC]: Received error (protocol code 2) during PASE process. Error CHIP:0x00000038
E (177587) chip[SC]: Failed during PASE session setup. Error CHIP:0x00000038
E (177597) chip[SVR]: Commissioning failed (attempt 5): Error CHIP:0x00000038

Note: Also tried to kill both chip tools and started fresh, but same issue was seen. 

Attached screen recording for the issue. 

SHA tested: 78776acf6f19eb72e63c98c1cad11054554b4091  (Feb 8th )

```

I (36637) wifi:connected with Desk1, aid = 2, channel 1, BW20, bssid = 4c:01:43:fa:78:26
I (36657) wifi:security: WPA2-PSK, phy: bgn, rssi: -25
I (36667) wifi:pm start, type: 1

I (36667) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (36677) chip[DL]: WIFI_EVENT_STA_CONNECTED
I (36677) chip[DL]: WiFi station state change: Connecting -> Connecting_Succeeded
W (36687) wifi:<ba-add>idx:0 (ifx:0, 4c:01:43:fa:78:26), tid:0, ssn:0, winSize:64
I (36687) chip[DL]: WiFi station state change: Connecting_Succeeded -> Connected
I (36707) chip[DL]: WiFi station interface connected
I (36707) chip[DIS]: Advertise operational node BFCE554D3ADFF563-000000000000000C
I (36717) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
E (36717) chip[SVR]: Operational advertising enabled
I (36727) app-devicecallbacks: Current free heap: 61212

I (36737) chip[IN]: Prepared secure message 0x3ffe8bb4 to 0xFFFFFFFB00000000 (1)  of type 0x9 and protocolId (0, 1) on exchange 31754r with MessageCounter:13621976.
I (36747) chip[IN]: Sending encrypted msg 0x3ffe8bb4 with MessageCounter:13621976 to 0xFFFFFFFB00000000 (1) at monotonic time: 35974 msec
I (36767) NimBLE: GATT procedure initiated: notify; 
I (36767) NimBLE: att_handle=14

I (36777) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (36777) chip[DL]: Done driving station state, nothing else to do...
I (36797) app-devicecallbacks: Current free heap: 61212

E (36797) chip[DL]: Long dispatch time: 118 ms, for event type 49152
I (36807) chip[DL]: Updating advertising data
I (36807) app-devicecallbacks: Current free heap: 62732

I (38097) chip[DL]: IP_EVENT_GOT_IP6
I (38097) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:baf0:09ff:fec4:9860
I (38107) app-devicecallbacks: Current free heap: 62688

I (38117) chip[DIS]: CHIP minimal mDNS started advertising.
I (38127) chip[DIS]: Advertise operational node BFCE554D3ADFF563-000000000000000C
I (38127) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (38147) chip[DIS]: Have operational credentials
I (38147) app-devicecallbacks: Current free heap: 62388

I (38297) ROUTE_HOOK: Received RIO
I (38297) ROUTE_HOOK: prefix FDE6:EEAC:B875:: lifetime 1800

I (38597) esp_netif_handlers: sta ip: 192.168.7.149, mask: 255.255.252.0, gw: 192.168.4.1
I (38597) chip[DL]: IP_EVENT_STA_GOT_IP
I (38597) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.7.149/255.255.252.0 gateway 192.168.4.1
I (38607) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (38627) app-devicecallbacks: Current free heap: 63060

[... 514 lines cut ...]
E (177327) chip[DL]: Long dispatch time: 2422 ms, for event type 3
I (177567) chip[EM]: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:1020627971 on exchange 57140r
E (177577) chip[SC]: Received error (protocol code 2) during PASE process. Error CHIP:0x00000038
E (177587) chip[SC]: Failed during PASE session setup. Error CHIP:0x00000038
E (177597) chip[SVR]: Commissioning failed (attempt 5): Error CHIP:0x00000038
I (177597) chip[DL]: NVS set: chip-factory/discriminator = 3840 (0xF00)
I (177617) chip[DIS]: CHIP minimal mDNS started advertising.
I (177637) chip[DIS]: Advertise operational node BFCE554D3ADFF563-000000000000000C
I (177647) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (177667) chip[DIS]: Have operational credentials
I (177667) chip[DIS]: Advertise commission parameter vendorID=9050 productID=65279 discriminator=3840/15
I (177677) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'.
I (177707) chip[IN]: Prepared unauthenticated message 0x3ffe8a04 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 57140r with MessageCounter:3887276535.
I (177717) chip[IN]: Sending unauthenticated msg 0x3ffe8a04 with MessageCounter:3887276535 to 0x0000000000000000 at monotonic time: 176944 msec
E (177737) chip[DL]: Long dispatch time: 168 ms, for event type 3

> I (185407) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (185507) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (215507) all-clusters-app: Display going to sleep...
I (299117) ROUTE_HOOK: Received RIO
I (299117) ROUTE_HOOK: prefix FDE6:EEAC:B875:: lifetime 1800

```
 #### Proposed Solution
Fix

dhrishi, 2022-02-16:
@woody-apple Is it possible to check if this works fine with chip-tool? It will help us to know whether it is a problem with iOS or M5/ESP32

cecille, 2022-02-25:
Tested the following with linux all-clusters and chip-tool

```
./out/debug/chip-tool pairing onnetwork-long 12345 20202021 3840
./out/debug/chip-tool administratorcommissioning open-basic-commissioning-window 100 12345 0 --timedInteractionTimeoutMs 10000
./out/debug/chip-tool pairing onnetwork-long 11111 20202021 3840
./out/debug/chip-tool basic read vendor-id 11111 0
./out/debug/chip-tool basic read vendor-id 12345 0
```
That works, so I think the basic 2nd admin pairing flow is OK. My M5 is acting up right now so I can't test there. Woody - are you using ECM or BCM to open the pairing window? There were some changes there recently that may be affecting your ability to get a second PASE connection. Have you tested recently on ToT?

cecille, 2022-03-07:
Assigning to Woody to answer the question above.

bzbarsky-apple, 2022-06-13:
If the pairing window was opened via iOS CHIPTool it's using ECM.  In which case scanning the original QR code will not work.  CHIPTool does output a manual code that can be used to pair with the device, but iirc does not show a scannable QR code.  So I assume the original QR code was scanned, in which case PASE failure is expected...

closed 2022-06-13 by bzbarsky-apple