project-chip/connectedhomeip #14981: Min/Max Reporting interval not updated with new values

kean-apple, 2022-02-09:
#### Problem
Minimal Reporting interval not updated with new values, keeps older values 

SHA: 983fedfcbc38909d73b80f8b5c1d2d8ac163d0ff
Initial: build iOS chip tool app

1. Launch iOS chip tool app
2. Pair M5/ESP32 board running all clusters app (IDF 4.4)
3. On chip tool app, go to temperature sensor & Set Reporting Min = 1s, Max = 2s & Delta = 1C
4. Change temperature setting on M5/ESP board eg. Device->Thermometer->External->Thermometer->Temp.  Verify reporting received in chip tool app & updated temp values show up on chip tool app after 1s
5. Set Reporting Min = 10s, Max = 20s & Delta = 2C
6. Change temperature setting on M5/ESP board eg. Device->Thermometer->External->Thermometer->Temp.  Verify reporting received in chip tool app & updated temp values show up on chip tool app after 10s

In Step 6, looks like updated intervals are not being respected. Reporting interval isn't set to 10s/20s. chip tool app is getting updates every 1/2 secs again.

M5/ESP32 logs:
//did receive updated 20s
I (135007) chip[IM]: Received status response, status is 0
I (135027) chip[DMG]: Refresh Subscribe Sync Timer with max 20 seconds
I (135027) chip[IN]: Prepared secure message 0x3ffca0d4 to 0x8C41E4976B03E5BF (1)  of type 0x4 and protocolId (0, 1) on exchange 22123r with MessageCounter:14038201.
I (135037) chip[IN]: Sending encrypted msg 0x3ffca0d4 with MessageCounter:14038201 to 0x8C41E4976B03E5BF (1) at monotonic time: 134264 msec
I (135217) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:9755680 on exchange 22123r
I (136447) ROUTE_HOOK: Received RIO
I (136447) ROUTE_HOOK: prefix FD1E:264F:85DD:: lifetime 1800
I (145027) chip[DMG]: Unblock report hold after min 10 seconds

//seems to revert back to 2 secs
on exchange 22124r
I (155487) chip[IM]: Received status response, status is 0
I (155507) chip[DMG]: Refresh Subscribe Sync Timer with max 2 seconds
I (155507) chip[IN]: Prepared secure message 0x3ffca0d4 to 0x8C41E4976B03E5BF (1)  of type 0x4 and protocolId (0, 1) on exchange 22124r with MessageCounter:14038205.
I (155517) chip[IN]: Sending encrypted msg 0x3ffca0d4 with MessageCounter:14038205 to 0x8C41E4976B03E5BF (1) at monotonic time: 154743 msec
I (155687) chip[EM]: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:9755684 on exchange 22124r
I (156507) chip[DMG]: Unblock report hold after min 1 seconds
I (157507) chip[DMG]: Refresh subscribe timer sync after 1 seconds

kean-apple, 2022-02-09:
[reporting-doesnt-update-interval-chiptool.txt](https://github.com/project-chip/connectedhomeip/files/8036208/reporting-doesnt-update-interval-chiptool.txt)
[reporting-doesnt-update-interval - m5 board.txt](https://github.com/project-chip/connectedhomeip/files/8036209/reporting-doesnt-update-interval.-.m5.board.txt)

mrjerryjohns, 2022-02-15:
This is likely because chip-tool is setting up a 2nd independent subscription as opposed to replacing the first one...

bzbarsky-apple, 2022-03-08:
Is this still an issue on current tip (after https://github.com/project-chip/connectedhomeip/pull/15943) with the wildcard reporting setup we're now using?

kean-apple, 2022-03-08:
Blocked from testing this due to https://github.com/project-chip/connectedhomeip/issues/15975

kean-apple, 2022-03-11:
Closing this issue. Looks fixed with https://github.com/project-chip/connectedhomeip/pull/15943. I do see subscriptions update with different timing intervals

closed 2022-03-11 by kean-apple