project-chip/connectedhomeip #14982: Unable to pair Thread matter accessory - crash

woody-apple, 2022-02-09:
#### Problem
Crash here:
```
Thread 3 Crashed::   Dispatch queue: com.apple.CoreHAP.CHIPAccessoryServer.clientQueue.4424842992
0   libsystem_kernel.dylib        	       0x210d81b98 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x24a3bc3bc pthread_kill + 268
2   libsystem_c.dylib             	       0x1e451b51c abort + 168
3   CHIP                          	       0x24c492684 chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets, 32u>::~EndPointManagerImplPool() + 164
4   CHIP                          	       0x24c49121c chip::Inet::EndPointManagerImplPool<chip::Inet::UDPEndPointImplSockets, 32u>::~EndPointManagerImplPool() + 32
5   libsystem_c.dylib             	       0x1e451be80 __cxa_finalize_ranges + 464
6   libsystem_c.dylib             	       0x1e45166fc exit + 32
7   XXXX                    	       0x24f9f4a10 -[XXX timerDidFire:] + 532
8   XXXX                  	       0x2102b05dc -[XXXX __handleExpiration] + 148
9   XXXX                  	       0x2102af128 ____activateIfNecessary_block_invoke + 40
10  libdispatch.dylib             	       0x1d90b7830 _dispatch_block_async_invoke2 + 148
11  libdispatch.dylib             	       0x1d90a8a2c _dispatch_client_callout + 20
12  libdispatch.dylib             	       0x1d90abeec _dispatch_continuation_pop + 500
13  libdispatch.dylib             	       0x1d90bf13c _dispatch_source_invoke + 1596
14  libdispatch.dylib             	       0x1d90b0000 _dispatch_lane_serial_drain + 376
15  libdispatch.dylib             	       0x1d90b0cb4 _dispatch_lane_invoke + 444
16  libdispatch.dylib             	       0x1d90bb500 _dispatch_workloop_worker_thread + 648
17  libsystem_pthread.dylib       	       0x24a3b60bc _pthread_wqthread + 288
18  libsystem_pthread.dylib       	       0x24a3b5e5c start_wqthread + 8
```
 #### Proposed Solution
Fix

bzbarsky-apple, 2022-06-13:
@woody-apple Are we still seeing this?

bzbarsky-apple, 2022-12-17:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/14859

closed 2022-12-17 by bzbarsky-apple