project-chip/connectedhomeip #14984: CHIP: chip::BitMapObjectPool<chip::app::TimedHandler, 8ul, (chip::OnObjectPoolDestruction)1>::~BitMapObjectPool

woody-apple, 2022-02-09:
#### Problem

Crash during shutdown
```
Thread 4 Crashed::   Dispatch queue: com.zigbee.chip.framework.controller.workqueue
0   libsystem_kernel.dylib        	       0x1d2608b98 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x20c9423dc pthread_kill + 272
2   libsystem_c.dylib             	       0x1a57d3538 abort + 168
3   CHIP                          	       0x10529ff6a chip::BitMapObjectPool<chip::app::TimedHandler, 8ul,>::~BitMapObjectPool() + 128
4   CHIP                          	       0x10529fdf6 chip::BitMapObjectPool<chip::app::TimedHandler, 8ul,>::~BitMapObjectPool() + 32
5   CHIP                          	       0x1052e9e22 chip::Controller::DeviceCommissioner::~DeviceCommissioner() + 584
6   CHIP                          	       0x1052df2de chip::Controller::DeviceCommissioner::~DeviceCommissioner() + 32
7   CHIP                          	       0x1052df30e chip::Controller::DeviceCommissioner::~DeviceCommissioner() + 32
8   CHIP                          	       0x10500df06 __32-[CHIPDeviceController shutdown]_block_invoke + 124
9   libdispatch.dylib             	       0x199dbae68 _dispatch_call_block_and_release + 32
10  libdispatch.dylib             	       0x199dbca2c _dispatch_client_callout + 20
11  libdispatch.dylib             	       0x199dc4124 _dispatch_lane_serial_drain + 668
12  libdispatch.dylib             	       0x199dc4c80 _dispatch_lane_invoke + 392
13  libdispatch.dylib             	       0x199dcf500 _dispatch_workloop_worker_thread + 648
14  libsystem_pthread.dylib       	       0x20c93c140 _pthread_wqthread + 288
15  libsystem_pthread.dylib       	       0x20c93bee0 start_wqthread + 8
```

 #### Proposed Solution
Fix

andy31415, 2022-02-09:
Note crash is mostly a symptom: it is saying that objects are still allocated by the time the pool destructor is run, so our code leaks things.

mrjerryjohns, 2022-02-15:
Any more details on repro @woody-apple ?

mrjerryjohns, 2022-03-08:
Closing given lack of more data.

closed 2022-03-08 by mrjerryjohns