project-chip/connectedhomeip #14987: Rename chip-tool-darwin to chip-tool

woody-apple, 2022-02-09:
As an aside, what is the difference beteween `chip-tool` and `chip-tool-darwin`? Both seem CLI. Would chip-tool not work on darwin as well? Why do both exist?

_Originally posted by @andy31415 in https://github.com/project-chip/connectedhomeip/issues/14816#issuecomment-1033827983_

bzbarsky-apple, 2022-02-10:
> As an aside, what is the difference beteween chip-tool and chip-tool-darwin?

@andy31415 One is a POSIX app that happens to run on Darwin (and Linux) and uses the C++ SDK APIs directly, and one is an Objective-C app that uses the Darwin Matter framework.

They both exists because the exercise different code paths and different API surfaces.

woody-apple, 2022-02-16:
Moving this out of v1.0, doesn't seem critical!

PR #18903 (2022-05-27): Rename chip-tool-darwin to darwin-framework-tool
#### Problem
chip-tool-darwin is confusing since there is a chip-tool that runs on Darwin. Rename chip-tool-darw to darwin-framework-tool to clarify that this tool is specifically for the use of CHIP.framework. 
- Fixes #14987

#### Change overview
- Renames chip-tool-darwin to darwin-framework-tool
- Removes a ZAP Readme file left.

#### Testing
- Re-ran regen_all_zap, no files were changed.
- Build
- Ran darwin-framework-tool commands.
closed 2022-05-27 by bzbarsky-apple