project-chip/connectedhomeip #14990: Seeing failure when tried to operate M5 light on 1st chip tool when we install 2nd chip tool app on the same device

vinay-apple, 2022-02-09:
#### Problem

Seeing failure when tried to operate M5 light on 1st chip tool when we install 2nd chip tool app on the same device 

Failure performing action. C++-mangled success callback type: 'PFvPvRKN4chip3app9DataModel14NullObjectTypeEE', error: ../../../../../../../../../../../CHIP-TOOL/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000041: No route to host

SHA: 983fedfcbc38909d73b80f8b5c1d2d8ac163d0ff

Steps to reproduce

 1. Install open source chip tool app on iOS device and pair with M5 board
 2. Toggle the light and see that it works
 3. Install a second instance of chip tool on the same iOS device
 4. Try to control the M5 board from 1st chip tool app and seeing failure on the chip tool logs. 

Expected:

Must be able to control with 1st chip tool app without issues when we have another instance of chip tool app installed on the same device

[Failed-toggle-when-2nd-chip-tool-installed-CHIPTool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8036942/Failed-toggle-when-2nd-chip-tool-installed-CHIPTool-logs.txt)
[Failed-toggle-when-2nd-chip-tool-installed-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8036943/Failed-toggle-when-2nd-chip-tool-installed-M5-logs.txt)
.

dhrishi, 2022-02-16:
@vinay-apple Is it possible to recreate the same through chip-tool. This will help us know if it is an issue with the iOS side or ESP32 side

vinay-apple, 2022-02-18:
@dhrishi the issue is seen with chip tool itself. When I have 1 instance of chip tool running I dont see issue with toggling M5 light. But as soon as I install second instance of chip tool (chip tool2) on the same iOS device we start to see the issue (No route to host) and unable to control M5 with the original chip tool (chip tool1) 

Note: I have not yet paired M5 with second instance of chip tool 2, just installed on the same device. 

Seeing the same issue even with recent commit 4d69db12bc061f6dd1408475e3969bf85f91b2e3

dhrishi, 2022-03-03:
>Is it possible to recreate the same through chip-tool.

@vinay-apple Sorry for the confusion. I meant trying `chip-tool` on Mac/Linux

vinay-apple, 2022-03-08:
@dhrishi will give this a try with chip-tool running in Mac. Meanwhile, on a recent SHA tested, we see the same issue. Added logs from M5 and chip tool. 

SHA Tested : c0373bbc244894909fb0acf55c73444e14088208

[no-route-to-host-M5-logs-SHA-c0373bbc244894909fb0acf55c73444e14088208.txt](https://github.com/project-chip/connectedhomeip/files/8201838/no-route-to-host-M5-logs-SHA-c0373bbc244894909fb0acf55c73444e14088208.txt)

[no-route-to-host-chiptool-logs-SHA-c0373bbc244894909fb0acf55c73444e14088208.txt](https://github.com/project-chip/connectedhomeip/files/8201837/no-route-to-host-chiptool-logs-SHA-c0373bbc244894909fb0acf55c73444e14088208.txt)

bzbarsky-apple, 2022-06-13:
Putting iOS CHIPTool in the background without quitting it will kill the socket, and then things are broken.

bzbarsky-apple, 2022-06-13:
Duplicate of #584

closed 2022-06-13 by bzbarsky-apple