project-chip/connectedhomeip #15003: Android chip-tool can't pair over only IPv6 , while linux chip-tool can do.

jamesluo11, 2022-02-10:
#### Problem
Android chip-tool can't pair over only IPv6 mode. i disable IPv4 mode in the device side.  While linux chip-tool can pair successfully with the device.

 #### Logs
**a.Logcat from android OPPO phone**
 
[logcat.txt](https://github.com/project-chip/connectedhomeip/files/8037443/logcat.txt)

**b.IP information on the android phone**
 OP4ED5:/ $ ip -6 route          **-->Nothing information to show**                                                                                                      
OP4ED5:/ $ 

OP4ED5:/ $ ip route                                                                                                                    
10.0.0.0/24 dev wlan0 proto kernel scope link src 10.0.0.15

OP4ED5:/ $ ip addr show wlan0                                                                                                          
38: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1420 qdisc mq state UP group default qlen 3000
    link/ether 4c:50:f1:e2:c5:51 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.15/24 brd 10.0.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::f864:e9f3:2292:9137/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever

**c.Log from device**
[device log.txt](https://github.com/project-chip/connectedhomeip/files/8037451/device.log.txt)

 #### Inital Analysis
While during CASE channel setup process, after device sends the Sigma2 msg, meanwhile sends several UDP packets to android chip-tool out, then device will report "Failed to Send CHIP message xx max retries: 3".  i can capture the wifi packets, device has sent out the packets, but don't receive the response from android chip-tool.
And as nothing router info from adb shell command "ip -6 route", so i guess route issue on the android system, udp packets are lost.

 #### Proposed Solution
I guess route issue on the android system
if the root cause is really the IPv6 route, is it possible chip-tool can add the default route ?

jamesluo11, 2022-02-10:
@tcarmelveilleux

mrcao2011, 2022-06-08:
the same issue：iOS chip-tool can't pair over only IPv6

mrcao2011, 2022-06-08:
> @tcarmelveilleux

the same issue：iOS chip-tool can't pair over only IPv6

tcarmelveilleux, 2022-08-12:
There is no route needed for link-local addresses.

Your DUT responded to IPv6 mDNS queries:
```
hip[DL] bk_read_data: chip-factory  serial-num failed

chip[DIS] Advertise commission parameter vendorID=9050 productID=65279 discriminator=3840/15
chip[DIS] CHIP minimal mDNS configured as 'Commissionable node device'.
chip[DIS] Broadcasting mDns reply for query from FE80::CA47:8CFF:FE42:914
chip[ZCL] DeviceEventCallback, event->Type:32784

chip[ZCL] Commissioning complete
chip[DL] StartAdvertising...
chip[DL] bk_read_data: chip-factory  discriminator failed

[gapm_cmp_evt_handler] conidx:0,operation:0xa9,status:0x0
chip[DL] cmd:1 idx:0 status:0

chip[DL] StartAdvertising...
chip[DL] StartAdvertising...
[gapm_cmp_evt_handler] conidx:0,operation:0xa4,status:0x0
chip[DL] cmd:3 idx:0 status:0
```

It even saw your phone's IPv6 address at some point:

```
chip[DIS] Broadcasting mDns reply for query from FE80::F864:E9F3:2292:9137
chip[DIS] QUERY IN/TXT: 613CE80016D9D594-0000000000000009._matter._tcp.local.
chip[DIS] Broadcasting mDns reply for query from FE80::F864:E9F3:2292:9137
chip[EM] Received message of type 0x30 with protocolId (0, 0) and MessageCounter:1653582002 on exchange 51603r
chip[EM] Handling via exchange: 51603r, Delegate: 0x40c574
chip[IN] CASE Server received Sigma1 message. Starting handshake. EC 0x40d7d8
```

The Sigma1 got received, Sigma2 was sent.

You have to debug your device and phone IPv6 link local communication *at the link layer*. This is not an Android chip-tool problem directly. We saw IPv6 traffic from phone and responded.

Assigning back to @jamesluo11

closed 2023-02-19 by stale[bot]