project-chip/connectedhomeip #15023: Followup: This seems pretty broken in that we will return a generic Failure here on the IM level instead of a useful "malformed message" error code.

yunhanw-google, 2022-02-10:
Followup: This seems pretty broken in that we will return a generic Failure here on the IM level instead of a useful "malformed message" error code.

This problem is not new with this PR, but please make sure there is an issue filed tracking it.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/14711#discussion_r803325458_

mrjerryjohns, 2022-06-23:
Is this really a spec issue? I don't see mention in the spec of the specific error that needs to be returned here...

bzbarsky-apple, 2022-06-27:
@mrjerryjohns Spec generally says that if action fields are invalid you return InvalidAction, no?

mrjerryjohns, 2022-07-13:
@yunhanw-google can you tackle this as well? It's in line with the other PR you already have in this space (#19356).

yunhanw-google, 2022-07-13:
yes

andy31415, 2022-07-18:
Is this required for v1.0? It looks like unfriendly message, however unsure about the "spec 1.0" tags. Is this a spec failure?

bzbarsky-apple, 2022-07-19:
> Is this a spec failure?

Technically, yes.  The spec defines what error code should be returned here, and we are returning the wrong one.

The "unfriendly message" is just a consequence of how the error code gets logged by the other side, but the wrong error code is the spec issue.

bzbarsky-apple, 2022-08-10:
This got fixed in https://github.com/project-chip/connectedhomeip/pull/21374 as far as I can tell.

closed 2022-08-10 by bzbarsky-apple