project-chip/connectedhomeip #15056: chip-tool unable to commission chip-app1 and all clusters app

wahajsyed, 2022-02-10:
#### Problem
chip-tool cannot commission all clusters app or chip-app on RaspberryPi+Linux with ToT

 #### Configuration:
OS: Ubuntu 21.10
Platform: RaspberryPi
CommitSHA: a08613bed0a7ed48efc96571d7a799590a2f3ed5
Setup: chip-tool and chip-app1/chip-all-clusters-app running on the same raspberry pi

<details>
<summary>Chip Tool Logs</summary>

```
ubuntu@ubuntu:~/chip-certification-tool/backend/third_party/connectedhomeip/repo/examples/chip-tool/out/debug$ ./chip-tool pairing onnetwork-long 0x654321 20202021 3840
[1644531046.749763][92866:92866] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-uBVhf5)
[1644531046.750478][92866:92866] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1644531046.750537][92866:92866] CHIP:DL: NVS set: chip-counters/reboot-count = 4 (0x4)
[1644531046.751401][92866:92866] CHIP:DL: Got Ethernet interface: eth0
[1644531046.751960][92866:92866] CHIP:DL: Found the primary Ethernet interface:eth0
[1644531046.752546][92866:92866] CHIP:DL: Got WiFi interface: wlan0
[1644531046.752618][92866:92866] CHIP:DL: Failed to reset WiFi statistic counts
[1644531046.752682][92866:92866] CHIP:IN: UDP::Init bind&listen port=5542
[1644531046.752790][92866:92866] CHIP:IN: UDP::Init bound to port=5542
[1644531046.752816][92866:92866] CHIP:IN: UDP::Init bind&listen port=5542
[1644531046.752903][92866:92866] CHIP:IN: UDP::Init bound to port=5542
[1644531046.752928][92866:92866] CHIP:IN: TransportMgr initialized
[1644531046.752963][92866:92866] CHIP:DIS: Init fabric pairing table with server storage
[1644531046.755051][92866:92866] CHIP:IN: Loading certs from storage
[1644531046.755181][92866:92866] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1644531046.755213][92866:92866] CHIP:IN: 0x04, 0x7e, 0x67, 0x39, 0x26, 0x44, 0x23, 0xa3, 
[1644531046.755237][92866:92866] CHIP:IN: 0x85, 0xb8, 0x19, 0xb5, 0xfd, 0xb8, 0x45, 0x60, 
[1644531046.755260][92866:92866] CHIP:IN: 0xef, 0x61, 0x7b, 0x06, 0xf3, 0x4c, 0x2c, 0x21, 
[1644531046.755282][92866:92866] CHIP:IN: 0xe7, 0x73, 0x20, 0x7a, 0x28, 0x04, 0xdf, 0xe4, 
[1644531046.755304][92866:92866] CHIP:IN: 0xab, 0xcf, 0xfc, 0xfe, 0x2e, 0x08, 0x9d, 0xa0, 
[1644531046.755327][92866:92866] CHIP:IN: 0x10, 0x2a, 0x0c, 0x4f, 0x1b, 0xcd, 0xdd, 0x8a, 
[1644531046.755350][92866:92866] CHIP:IN: 0x62, 0x0f, 0xf3, 0x4a, 0xd5, 0x42, 0x03, 0xc2, 
[1644531046.755372][92866:92866] CHIP:IN: 0x69, 0x78, 0xca, 0xbd, 0x29, 0x31, 0x70, 0x33, 
[1644531046.755391][92866:92866] CHIP:IN: 0x9a, 
[1644531046.755452][92866:92866] CHIP:IN: Generated compressed fabric ID
[1644531046.755478][92866:92866] CHIP:IN: 0x37, 0x49, 0x22, 0xd7, 0xb8, 0x89, 0x72, 0x88, 
[1644531046.755752][92866:92866] CHIP:IN: Loading certs from storage
[1644531046.755830][92866:92866] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1644531046.755859][92866:92866] CHIP:IN: 0x04, 0x7e, 0x67, 0x39, 0x26, 0x44, 0x23, 0xa3, 
[1644531046.755883][92866:92866] CHIP:IN: 0x85, 0xb8, 0x19, 0xb5, 0xfd, 0xb8, 0x45, 0x60, 
[1644531046.755906][92866:92866] CHIP:IN: 0xef, 0x61, 0x7b, 0x06, 0xf3, 0x4c, 0x2c, 0x21, 
[1644531046.755928][92866:92866] CHIP:IN: 0xe7, 0x73, 0x20, 0x7a, 0x28, 0x04, 0xdf, 0xe4, 
[1644531046.755950][92866:92866] CHIP:IN: 0xab, 0xcf, 0xfc, 0xfe, 0x2e, 0x08, 0x9d, 0xa0, 
[1644531046.755972][92866:92866] CHIP:IN: 0x10, 0x2a, 0x0c, 0x4f, 0x1b, 0xcd, 0xdd, 0x8a, 
[1644531046.755995][92866:92866] CHIP:IN: 0x62, 0x0f, 0xf3, 0x4a, 0xd5, 0x42, 0x03, 0xc2, 
[1644531046.756017][92866:92866] CHIP:IN: 0x69, 0x78, 0xca, 0xbd, 0x29, 0x31, 0x70, 0x33, 
[1644531046.756036][92866:92866] CHIP:IN: 0x9a, 
[1644531046.756077][92866:92866] CHIP:IN: Generated compressed fabric ID
[1644531046.756103][92866:92866] CHIP:IN: 0xae, 0x15, 0x97, 0x2c, 0x47, 0x66, 0xa3, 0xb7, 
[1644531046.756321][92866:92866] CHIP:IN: Loading certs from storage
[1644531046.756392][92866:92866] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1644531046.756422][92866:92866] CHIP:IN: 0x04, 0x7e, 0x67, 0x39, 0x26, 0x44, 0x23, 0xa3, 
[1644531046.756445][92866:92866] CHIP:IN: 0x85, 0xb8, 0x19, 0xb5, 0xfd, 0xb8, 0x45, 0x60, 
[1644531046.756468][92866:92866] CHIP:IN: 0xef, 0x61, 0x7b, 0x06, 0xf3, 0x4c, 0x2c, 0x21, 
[1644531046.756490][92866:92866] CHIP:IN: 0xe7, 0x73, 0x20, 0x7a, 0x28, 0x04, 0xdf, 0xe4, 
[1644531046.756512][92866:92866] CHIP:IN: 0xab, 0xcf, 0xfc, 0xfe, 0x2e, 0x08, 0x9d, 0xa0, 
[1644531046.756534][92866:92866] CHIP:IN: 0x10, 0x2a, 0x0c, 0x4f, 0x1b, 0xcd, 0xdd, 0x8a, 
[1644531046.756557][92866:92866] CHIP:IN: 0x62, 0x0f, 0xf3, 0x4a, 0xd5, 0x42, 0x03, 0xc2, 
[1644531046.756579][92866:92866] CHIP:IN: 0x69, 0x78, 0xca, 0xbd, 0x29, 0x31, 0x70, 0x33, 
[1644531046.756598][92866:92866] CHIP:IN: 0x9a, 
[1644531046.756637][92866:92866] CHIP:IN: Generated compressed fabric ID
[1644531046.756662][92866:92866] CHIP:IN: 0xa4, 0xf8, 0x62, 0xe1, 0x81, 0xa3, 0xf6, 0xd7, 
[1644531046.756984][92866:92866] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-L4lIwY)
[1644531046.771065][92866:92866] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1644531046.771135][92866:92866] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 4000 (0xFA0)
[1644531046.771287][92866:92866] CHIP:ZCL: Using ZAP configuration...
[... 408 lines cut ...]
[1644531047.096777][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::781a:3cff:fe77:3e48 on port 52033
[1644531047.100077][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::781a:3cff:fe77:3e48 on port 52033
[1644531047.103519][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.107269][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.110478][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.113797][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.117348][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.121459][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.125664][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.128874][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.132788][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.135857][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::e88f:69ff:fe8a:df53 on port 45506
[1644531047.139280][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.142456][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.145584][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.148864][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.152224][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.155766][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.158972][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.162149][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.165505][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
[1644531047.168733][92061:92061] CHIP:DIS: Directly sending mDns reply to peer fe80::d80b:10ff:fe11:35d9 on port 43077
```

</details>

bzbarsky-apple, 2022-02-10:
@wahajsyed can you bisect to figure out what commit introduced the problem?  In general onnetwork commissioning on Linux works (it's tested in CI all the time), so presumably there is some difference between your setup and what CI is doing.

jamesluo11, 2022-02-11:
I also encounter this issue. Below is device's log

chip[IN] BLE EndPoint 0x410b98 Connection Complete
chip[DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
chip[DL] _OnPlatformEvent default:  event->Type = 32778
chip[ZCL] DeviceEventCallback, event->Type:32778

chip[ZCL] CHIPoBLE connection established
chip[DL] write_cb:conn_idx:0, prf_id:0, add_id:2, len:75, data[0]:0d

chip[DL] _OnPlatformEvent kCHIPoBLEWriteReceived
**chip[IN] Received malformed unsecure packet with source 0x0000000000000000 destination 0x0000000000000000**

bzbarsky-apple, 2022-02-11:
@jamesluo11 Again, can you bisect?

jamesluo11, 2022-02-11:
@bzbarsky-apple yeah,  i just upgrade the android chip-tool, the issue was gone. So the root cause is there was a mismatch between node and controller.

wahajsyed, 2022-02-11:
@bzbarsky-apple - I think mine was a version mismatch too, chip-app inside docker was on the older version. Clearing out the build directories, env and rebuilding binaries and docker containers fixed it.

closed 2022-02-11 by wahajsyed