project-chip/connectedhomeip #15059: ESP32: Reporting stopped working after changing temp sensor once

kean-apple, 2022-02-10:
After configuring reporting on the ESP32 board, Reporting stopped working after changing temp sensor once 

1. build iOS chip tool app
2. Pair M5/ESP32 running all clusters app (IDF v4.4)
3. Go to temperature sensor & enable reporting - Min: 1s, Max: 2s, Delta: 1C
4. On M5 board, change temp sensor values (Devices->Thermometer->External->Thermometer->Temp) eg. 21C-> 30C
5. On Chiptool app, verify that temp values are changing

In step 5, temp sensor moved from 21->22C then stopped showing any further updates

I seen this sometimes but when in the state its not recoverable. 

[reporting-worked-once-to-22-m5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8044862/reporting-worked-once-to-22-m5-logs.txt)
[reporting-worked-once-to-22-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8044863/reporting-worked-once-to-22-chiptool-logs.txt)

M5 logs:
M]: Received message of type 0x1 with protocolId (0, 1) and MessageCounter:7228739 on exchange 10215i
E (66968) chip[EM]: Found matching exchange: 10215i, Delegate: 0x3ffcf8a8
E (66988) chip[EM]: Rxd Ack; Removing MessageCounter:11807707 from Retrans Table on exchange 10215i
E (66988) chip[EM]: Removed CHIP MessageCounter:11807707 from RetransTable on exchange 10215i
E (66998) chip[IM]: Received status response, status is 0
E (67018) chip[DMG]: <RE> OnReportConfirm: NumReports = 0
E (67018) chip[DMG]: IM RH moving to [GeneratingReports]
E (67018) chip[EM]: Sending Standalone Ack for MessageCounter:7228739 on exchange 10215i
E (67028) chip[IN]: Prepared secure message 0x3ffe8394 to 0xDBB3E4A04234F04F (1)  of type 0x10 and protocolId (0, 0) on exchange 10215i with MessageCounter:11807708.
E (67048) chip[IN]: Sending encrypted msg 0x3ffe8394 with MessageCounter:11807708 to 0xDBB3E4A04234F04F (1) at monotonic time: 66267 msec
E (67058) chip[EM]: Flushed pending ack for MessageCounter:7228739 on exchange 10215i
E (67068) chip[DL]: Long dispatch time: 103 ms, for event type 3
I (67148) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (67448) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (67448) all-clusters-app: editing attribute as integer: 22 (+)
E (67458) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d932
I (67458) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (67468) app-devicecallbacks: Unhandled cluster ID: 1026
I (67488) app-devicecallbacks: Current free heap: 62268

I (67678) all-clusters-app: Display awake but will switch off automatically in 30 seconds
E (67888) chip[DMG]: Unblock report hold after min 1 seconds
I (68378) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (68378) all-clusters-app: editing attribute as integer: 23 (+)
E (68388) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d933
I (68388) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (68398) app-devicecallbacks: Unhandled cluster ID: 1026
I (68418) app-devicecallbacks: Current free heap: 62268

I (68608) all-clusters-app: Display awake but will switch off automatically in 30 seconds
E (68898) chip[DMG]: Refresh subscribe timer sync after 1 seconds
I (69058) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (69058) all-clusters-app: editing attribute as integer: 24 (+)
E (69068) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d934
I (69068) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (69088) app-devicecallbacks: Unhandled cluster ID: 1026
I (69098) app-devicecallbacks: Current free heap: 62268

//Don't see the 1s subscriber timer anymore ..chiptool app doesn't update values either
I (69288) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (69788) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (69788) all-clusters-app: editing attribute as integer: 25 (+)
E (69798) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d935
I (69798) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (69808) app-devicecallbacks: Unhandled cluster ID: 1026
I (69828) app-devicecallbacks: Current free heap: 62268

I (70018) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (70518) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (70518) all-clusters-app: editing attribute as integer: 26 (+)
E (70528) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d936
I (70528) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (70538) app-devicecallbacks: Unhandled cluster ID: 1026
I (70558) app-devicecallbacks: Current free heap: 62268

I (70748) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (71348) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (71348) all-clusters-app: editing attribute as integer: 27 (+)
E (71358) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d937
I (71358) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (71368) app-devicecallbacks: Unhandled cluster ID: 1026
I (71388) app-devicecallbacks: Current free heap: 62268

I (71578) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (72328) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (72328) all-clusters-app: editing attribute as integer: 28 (+)
E (72338) chip[DMG]: Endpoint 1, Cluster 0x0000_0402 update version to d370d938
I (72338) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (72348) app-devicecallbacks: Unhandled cluster ID: 1026
I (72368) app-devicecallbacks: Current free heap: 62268

jadhavrohit924, 2022-02-15:
With the above steps using chip-tool I got messages in chip-tool logs when I changed measured-value on the latest codebase (commit: f05070aed57f941533f13b6a1d9b6871b1f930d7)

kean-apple, 2022-02-15:
This seems to be working better in the past 2 days

dhrishi, 2022-02-16:
@kean-apple Please close the issue if this works fine