project-chip/connectedhomeip #15089: Media Playback Cluster-[TC-MC-6.3]-Step 2 fails to reads the SampledPosition attribute.

Rajashreekalmane, 2022-02-11:
#### Problem
Running the command "./chip-tool mediaplayback read position 1 3" does not read the SampledPosition attribute and also it's not same as the expected outcome.

 [TC-MC-6.3] Advanced Seek Media Playback Verification.
[image]

PFA Test log:
[[TC-MC-6.3].txt](https://github.com/project-chip/connectedhomeip/files/8049921/TC-MC-6.3.txt)

lazarkov, 2022-02-17:
It's now:
`out/debug/standalone/chip-tool mediaplayback read sampled-position 1 3`

lazarkov, 2022-02-17:
Renamed from position to sampled-position.

Here's the response I've got:
`[1645110122661] [60169:1649264] CHIP: [TOO] Endpoint: 3 Cluster: 0x0000_0506 Attribute 0x0000_0003DataVersion: 2022448471
[1645110122661] [60169:1649264] CHIP: [TOO]   SampledPosition: {
[1645110122661] [60169:1649264] CHIP: [TOO]     UpdatedAt: 0
[1645110122661] [60169:1649264] CHIP: [TOO]     Position: 0
[1645110122662] [60169:1649264] CHIP: [TOO]    }`

closed 2022-02-17 by lazarkov

lazarkov, 2022-02-24:
@Rajashreekalmane should be good now?