project-chip/connectedhomeip #15120: Reject Write Requests if one is already outstanding

mrjerryjohns, 2022-02-12:
On the server, we should reject Write requests if one is already outstanding. This can only happen if the outstanding write is chunked and more messages to convey the remaining data are forthcoming on the exchange.

Ideally, we would achieve serialization at the attribute level, but that isn't easy to achieve.

bzbarsky-apple, 2022-02-12:
Do we have non-scoped lists that are writable?  We should check...

Because if not, we could restrict this to "outstanding write on the same fabric"...

msandstedt, 2022-02-12:
Isn't this the same as supporting only one write handler at a time?

It seems concerning to allow any client to be granted a global write on any node.  Can't the lock be held for a very long time?  Should we do the opposite and cancel pending writes if another is initiated?

mrjerryjohns, 2022-02-15:
> Because if not, we could restrict this to "outstanding write on the same fabric"...

That feels flimsy to do that predicated on the current set of data model features for v1.0. This hack will then get backed in and be difficult to unravel later.

mrjerryjohns, 2022-02-15:
> Should we do the opposite and cancel pending writes if another is initiated?

That can leave the target in a partial state since the new write from a different sender is likely destined to a different attribute.

mrjerryjohns, 2022-02-15:
> Isn't this the same as supporting only one write handler at a time?

Good point. Yes, I think it's equivalent.

> Can't the lock be held for a very long time

Not sure what long would be, but since the server would time out the exchange fairly quickly (since the CRMP retries would be done at the active interval).

bzbarsky-apple, 2022-02-15:
> This hack will then get baked in and be difficult to unravel later.

Well, ideally we would, post-1.0 do something more fine-grained than "only one writer of a list allowed".

I feel like the failure mode here (writes failing for potentially a double-digit number of seconds) is serious enough we should try to minimize the impact.

erjiaqing, 2022-02-24:
One approach is to record last processed path (EP/C/A, no list index) in the WriteHandler, when we are processing another attribute write request, we check if we are writing the same attribute in other write transactions.

If we found potential conflicts, we will reject the data with IM::Busy status code.

This should work since all of this will happen in the mainloop, and chunked list won't be breaked by other attributes.

This will increase the throughput for write transactions.

We really don't have to limit the write transactions in parallel to 1 when the resource is enough and we won't modify the same attribute for most cases.

erjiaqing, 2022-02-25:
Mistakenly attached wrong PR to this, the PR for this issue is still WIP

PR #15623 (2022-03-01): [IM] Reject conflict WriteRequests
#### Problem

Fixes #15120

Reject Write Requests if one is already outstanding, when the paths are being written by another write transaction.

#### Change overview

- `InteractionModel::Status::Busy` will be returned in this case.

#### Testing

Updated `src/controller/tests/TestWriteChunking.cpp` to cover this
closed 2022-03-01 by andy31415