project-chip/connectedhomeip #15132: Wrong ID number in CertificateChainResponse during Commissioning Process

wissn84, 2022-02-14:
#### Problem
 * expected behavior
In spec CertificateChainResponse (11.18.7.4) id is 1, But in chip-tool response is 0

 * actual behavior
CertificateChainResponse in chip-tool response "CommandData is 1" 
[image]

 * screenshots, images, logs, etc.

Log in chip-tool is below
```
[1644801185.496007][255601:255606] CHIP:EM: Received message of type 0x9 with protocolId (0, 1) and MessageCounter:13653466 on exchange 36606i
[1644801185.496013][255601:255606] CHIP:EM: Found matching exchange: 36606i, Delegate: 0x7f82b4005db0
[1644801185.496020][255601:255606] CHIP:EM: Rxd Ack; Removing MessageCounter:5874329 from Retrans Table on exchange 36606i
[1644801185.496024][255601:255606] CHIP:EM: Removed CHIP MessageCounter:5874329 from RetransTable on exchange 36606i
[1644801185.496031][255601:255606] CHIP:DMG: ICR moving to [ResponseRe]
[1644801185.496040][255601:255606] CHIP:DMG: InvokeResponseMessage =
[1644801185.496044][255601:255606] CHIP:DMG: {
[1644801185.496049][255601:255606] CHIP:DMG:    suppressResponse = false, 
[1644801185.496051][255601:255606] CHIP:DMG:    InvokeResponseIBs =
[1644801185.496061][255601:255606] CHIP:DMG:    [
[1644801185.496065][255601:255606] CHIP:DMG:            InvokeResponseIB =
[1644801185.496072][255601:255606] CHIP:DMG:            {
[1644801185.496076][255601:255606] CHIP:DMG:                    CommandDataIB =
[1644801185.496081][255601:255606] CHIP:DMG:                    {
[1644801185.496086][255601:255606] CHIP:DMG:                            CommandPathIB =
[1644801185.496091][255601:255606] CHIP:DMG:                            {
[1644801185.496096][255601:255606] CHIP:DMG:                                    EndpointId = 0x0,
[1644801185.496101][255601:255606] CHIP:DMG:                                    ClusterId = 0x3e,
[1644801185.496106][255601:255606] CHIP:DMG:                                    CommandId = 0x3,
[1644801185.496111][255601:255606] CHIP:DMG:                            },
[1644801185.496117][255601:255606] CHIP:DMG: 
[1644801185.496122][255601:255606] CHIP:DMG:                            CommandData = 
[1644801185.496127][255601:255606] CHIP:DMG:                            {
[1644801185.496132][255601:255606] CHIP:DMG:                                    0x0 = [
[1644801185.496137][255601:255606] CHIP:DMG:                                            ... (byte string too long) ...
[1644801185.496142][255601:255606] CHIP:DMG:                            ]
[1644801185.496147][255601:255606] CHIP:DMG:                            },
[1644801185.496151][255601:255606] CHIP:DMG:                    },
[1644801185.496158][255601:255606] CHIP:DMG: 
[1644801185.496162][255601:255606] CHIP:DMG:            },
[1644801185.496169][255601:255606] CHIP:DMG: 
[1644801185.496173][255601:255606] CHIP:DMG:    ],
[1644801185.496179][255601:255606] CHIP:DMG: 
[1644801185.496184][255601:255606] CHIP:DMG:    InteractionModelRevision = 1
[1644801185.496187][255601:255606] CHIP:DMG: },

```
[chip-tool-controller-2:14.txt](https://github.com/project-chip/connectedhomeip/files/8058460/chip-tool-controller-2.14.txt)

 #### Proposed Solution

cecille, 2022-02-25:
I think this might just be a typo in the spec - there's only one returned field for that command, there's no reason for it to be ID 1.
I put up a PR here: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/4984.

cecille, 2022-02-28:
@wissn84 - if we change that value in the spec, is there a corresponding change that needs to happen in the tests (ie, is this feedback from the CSG or just something you noticed and reported on your own)?

cecille, 2022-03-31:
Spec changed in https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/4984

cecille, 2022-03-31:
@jeonghwan7 - was this issue from a CSG test? The spec was updated, just trying to figure out if there's also a CSG test that needs to be updated.

bzbarsky-apple, 2022-05-21:
Fixed in spec, not SDK issue here.

closed 2022-05-21 by bzbarsky-apple