project-chip/connectedhomeip #15134: Commissioning over BLE and IP are failing Bridge-app.

sowmyassp, 2022-02-14:
#### Problem
Pairing  DUT with chip-tool is failing on Bridge-app (CommitID - 6380d38f44c46f9e8f684024bf99c6a203622455) on Master Branch

thok-itx, 2022-02-14:
Also reported within following issue: https://github.com/project-chip/connectedhomeip/issues/15050

sowmyassp, 2022-02-21:
@PSONALl - I tried with commit ID (d671a523745079b1ea1778ee4885159868d9c002) which you have mentioned but still facing same issue.

sowmyassp, 2022-03-15:
I can't launch bridge-app with commitID (2982c0ca0623773327def4cc7180d07f6b959c42) 
This is the issue I am facing 
[image]

Attached log:
[bridge_launching.txt](https://github.com/project-chip/connectedhomeip/files/8250434/bridge_launching.txt)

sowmyassp, 2022-04-11:
Still facing same issue with commitID (fa5561c23d8c1f9464da8669931affbae0660f51)

PR #17694 (2022-04-25): Fix bridge-app init
#### Problem
- Bridge app did not have proper CommissionableDataProvider
  init and nobody caught it until some testing weeks later.

Fixes #15134

#### Change overview
- This PR adds the necessary init calls and also refactors
  elements of the bridge app structure to prevent this from
  happening again.

#### Testing
- All unit tests pass
- Cert tests pass
- chip-bridge-app works and is commissionablen by chip-tool and chip-repl
commit 5ecc0b3c (2022-04-25): Fix bridge-app init (#17694)

* Fix bridge-app init

- Bridge app did not have proper CommissionableDataProvider
  init and nobody caught it until some testing weeks later.
- This PR adds the necessary init calls and also refactors
  elements of the bridge app structure to prevent this from
  happening again.

Fixes #15134

Testing done:
- All unit tests pass
- Cert tests pass
- chip-bridge-app works and is commissionablen by chip-tool and chip-repl

* Fix CI
closed 2022-04-25 by andy31415