project-chip/connectedhomeip #15146: Commissioning over BLE fails on iOS 15.3 and macOS 12.2.1

dtodor, 2022-02-14:
Commissioning over BLE doesn't seem to work with the current code base. I've tested the following setups:

Controllers
- macOS chip-tool (macOS versions 12.2.0 and 12.2.1)
- macOS python (macOS versions 12.2.0 and 12.2.1)
- iOS CHIP Tool App (iOS version 15.3)

Devices
- EFR32 window-app
- ESP32 all-clusters-app

Here is the output of the iOS app:

```
2022-02-14 15:14:30.270653+0100 CHIPTool[12159:4753617] CHIPOperationalCredentialsDelegate: No NOC Signer provided, using self managed keys
2022-02-14 15:14:30.276743+0100 CHIPTool[12159:4753617] Found an existing self managed keypair in the keychain
2022-02-14 15:14:30.276817+0100 CHIPTool[12159:4753617] Deserializing the key
2022-02-14 15:14:30.277120+0100 CHIPTool[12159:4753617] CHIPOperationalCredentialsDelegate::init returning ../../../../../../../../../../../Downloads/connectedhomeip.git/src/crypto/CHIPCryptoPALmbedTLS.cpp:761: Success
2022-02-14 15:14:30.277175+0100 CHIPTool[12159:4753617] PersistentStorageDelegate Sync Get Value for Key: com.zigbee.chip.commissioner.device_id
2022-02-14 15:14:30.277288+0100 CHIPTool[12159:4753617] CHIPPersistentStorageDelegate Get Value for Key: com.zigbee.chip.commissioner.device_id, value MZ0Y2ldcHyc=
2022-02-14 15:14:30.277332+0100 CHIPTool[12159:4753617] Found 271f5c57da189d31 node ID for the controller
2022-02-14 15:14:30.294783+0100 CHIPTool[12159:4753617] PersistentStorageDelegate Sync Get Value for Key: MatterCARootCert0
2022-02-14 15:14:30.294891+0100 CHIPTool[12159:4753617] CHIPPersistentStorageDelegate Get Value for Key: MatterCARootCert0, value MIIB2jCCAYCgAwIBAgIBADAKBggqhkjOPQQDAjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwHhcNMjIwMjE0MTUxMjA0WhcNMjMwMjE0MTUxMjA0WjBEMSAwHgYKKwYBBAGConwBBAwQMDAwMDAwMDAwMDAwMDREMjEgMB4GCisGAQQBgqJ8AQUMEDAwMDAwMDAwMDAwMDAwMDAwWTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAARkLVBBkyfWjyfeaSHkpxiMHmOY/DF8bqfBVflJnPQgfk+IXDFXwFB/KSIbTFAxLMDoOgHqKmu1gsw3tEaJLEM9o2MwYTAPBgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBBjAdBgNVHQ4EFgQUYOEPuKIBPDjFonjgWRDqowCyMKIwHwYDVR0jBBgwFoAUYOEPuKIBPDjFonjgWRDqowCyMKIwCgYIKoZIzj0EAwIDSAAwRQIgR75fUq30PfbeefJOpn9XZzvfW4wAupJGbAFCuBPAk4ECIQC9XqRS0259UN04c8si+ukV7AaEjnhldLRyRo2nj4A8Mw==
2022-02-14 15:14:30.295363+0100 CHIPTool[12159:4753617] [all] 🔵 [1644848070295] [12159:4753617] CHIP: [DL] Found user documents directory: file:///var/mobile/Containers/Data/Application/F79BEE14-BE33-4459-BCDB-E7ED986049EC/Documents/
2022-02-14 15:14:30.295419+0100 CHIPTool[12159:4753617] [all] 🔵 [1644848070295] [12159:4753617] CHIP: [DL] KVS will be written to: file:///var/mobile/Containers/Data/Application/F79BEE14-BE33-4459-BCDB-E7ED986049EC/Documents/chip.store
2022-02-14 15:14:30.302327+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [DL] _Init
2022-02-14 15:14:30.302421+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [IN] UDP::Init bind&listen port=0
2022-02-14 15:14:30.302543+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [IN] UDP::Init bound to port=59058
2022-02-14 15:14:30.302570+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [IN] UDP::Init bind&listen port=0
2022-02-14 15:14:30.302626+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [IN] UDP::Init bound to port=52379
2022-02-14 15:14:30.302693+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [IN] TransportMgr initialized
2022-02-14 15:14:30.302744+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070302] [12159:4753617] CHIP: [DIS] Init fabric pairing table with server storage
2022-02-14 15:14:30.302796+0100 CHIPTool[12159:4753617] PersistentStorageDelegate Sync Get Value for Key: F01/Fabric1
2022-02-14 15:14:30.302838+0100 CHIPTool[12159:4753617] CHIPPersistentStorageDelegate Get Value for Key: F01/Fabric1, value AQAAAAAAAAAxnRjaV1wfJwAAAAAAAAAAAADvAAAA/AAECqLHqK9ufybRQxZNhOndYSxDWXzXexzTCKxD7gZM/1y4pdw01tltQ/ir8KfAP1Gv1666iAeNFuGv+5zOSRUX7srVu+JLQTQIrQuQAhU/y47m3ireZGO/bMlmcniabrunAAAAAAAAAGEAAAAAAAAAFTABAQAkAgE3AyUU0gQkFQAYJgTELZ0pJgVEYX4rNwYlFNIEJBUAGCQHASQIATAJQQRkLVBBkyfWjyfeaSHkpxiMHmOY/DF8bqfBVflJnPQgfk+IXDFXwFB/KSIbTFAxLMDoOgHqKmu1gsw3tEaJLEM9Nwo1ASkBGCQCYDAEFGDhD7iiATw4xaJ44FkQ6qMAsjCiMAUUYOEPuKIBPDjFonjgWRDqowCyMKIYMAtAR75fUq30PfbeefJOpn9XZzvfW4wAupJGbAFCuBPAk4G9XqRS0259UN04c8si+ukV7AaEjnhldLRyRo2nj4A8MxgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAVMAEBASQCATcDJRTSBCQVABgmBMQtnSkmBURhfis3BiQVACcRMZ0Y2ldcHycYJAcBJAgBMAlBBAqix6ivbn8m0UMWTYTp3WEsQ1l813sc0wisQ+4GTP9cuKXcNNbZbUP4q/CnwD9Rr9euuogHjRbhr/uczkkVF+43CjUBKAEYJAIBNgMEAgQBGDAEFKMKhVx3hHDwFA0hRdYd4to13+EsMAUUYOEPuKIBPDjFonjgWRDqowCyMKIYMAtANIk0uBaAPpyGx1/OzsXRO+Ag/GDQ/yzfftC8/K0qtEDuj03Be1gomR7j95supkcXNQN0YFWidw1tcQKc99UX/BgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
2022-02-14 15:14:30.303135+0100 CHIPTool[12159:4753617] [all] 🔵 [1644848070303] [12159:4753617] CHIP: [IN] Loading certs from storage
2022-02-14 15:14:30.303260+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000000 and root pubkey
2022-02-14 15:14:30.303285+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x04, 0x64, 0x2d, 0x50, 0x41, 0x93, 0x27, 0xd6,
2022-02-14 15:14:30.303343+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x8f, 0x27, 0xde, 0x69, 0x21, 0xe4, 0xa7, 0x18,
2022-02-14 15:14:30.303364+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x8c, 0x1e, 0x63, 0x98, 0xfc, 0x31, 0x7c, 0x6e,
2022-02-14 15:14:30.303383+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0xa7, 0xc1, 0x55, 0xf9, 0x49, 0x9c, 0xf4, 0x20,
2022-02-14 15:14:30.303402+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x7e, 0x4f, 0x88, 0x5c, 0x31, 0x57, 0xc0, 0x50,
2022-02-14 15:14:30.303420+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x7f, 0x29, 0x22, 0x1b, 0x4c, 0x50, 0x31, 0x2c,
2022-02-14 15:14:30.303438+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0xc0, 0xe8, 0x3a, 0x01, 0xea, 0x2a, 0x6b, 0xb5,
2022-02-14 15:14:30.303648+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x82, 0xcc, 0x37, 0xb4, 0x46, 0x89, 0x2c, 0x43,
2022-02-14 15:14:30.303671+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x3d,
2022-02-14 15:14:30.303733+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] Generated compressed fabric ID
2022-02-14 15:14:30.303754+0100 CHIPTool[12159:4753617] [all] 🟢 [1644848070303] [12159:4753617] CHIP: [IN] 0x0a, 0x99, 0x7f, 0xfc, 0xaf, 0x76, 0x49, 0xf6,
2022-02-14 15:14:30.303778+0100 CHIPTool[12159:4753617] PersistentStorageDelegate Sync Get Value for Key: F02/Fabric2
2022-02-14 15:14:30.303806+0100 CHIPTool[12159:4753617] CHIPPersistentStorageDelegate Get Value for Key: F02/Fabric2, value (null)
2022-02-14 15:14:30.303834+0100 CHIPTool[12159:4753617] PersistentStorageDelegate Sync Get Value for Key: F03/Fabric3
2022-02-14 15:14:30.303862+0100 CHIPTool[12159:4753617] CHIPPersistentStorageDelegate Get Value for Key: F03/Fabric3, value (null)
2022-02-14 15:14:30.303885+0100 CHIPTool[12159:4753617] PersistentStorageDelegate Sync Get Value for Key: F04/Fabric4
2022-02-14 15:14:30.303908+0100 CHIPTool[12159:4753617] CHIPPersistentStorageDelegate Get Value for Key: F04/Fabric4, value (null)
[... cut ...]
I (13457) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (13457) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (13457) wifi:Haven't to connect to a suitable AP now!
I (13467) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
I (13467) chip[DL]: Next WiFi station reconnect in 5000 ms
I (13477) chip[DL]: Done driving station state, nothing else to do...
I (13487) app-devicecallbacks: Current free heap: 89092
```

It seems that the BLE connection is established, but the service discovery fails.

dhrishi, 2022-02-15:
>It seems that the BLE connection is established, but the service discovery fails.

Service discovery fails as the device is not able to successfully connect to the Wi-Fi network. See the logs:
```
I (5287) chip[DL]: WiFi station state change: Connecting -> Connecting_Failed
W (5287) wifi:Haven't to connect to a suitable AP now!
I (5297) chip[DL]: WiFi station state change: Connecting_Failed -> NotConnected
```
Can you please confirm if the AP SSID/passphrase is correct or the AP is in range?

dtodor, 2022-02-15:
The BLE service discovery fails:

```
2022-02-14 15:14:34.455678+0100 CHIPTool[12159:4753756] [all] 🔴 [1644848074455] [12159:4753756] CHIP: [BLE] Service not found on the device.
```

The WiFi data is configured in the app, the WiFi credentials are not being provisioned. Not even a PASE session is established.

dhrishi, 2022-02-16:
The device side logs show a Wi-Fi STA connection being initiated and unsuccessful, and may be that's why the BLE related stuff isn't initialized. If the regular BLE Wi-Fi commissioning is to be carried out from the iOS CHIPTool app, then can you please erase-flash on the ESP32 side and try again

dtodor, 2022-02-16:
After erasing the flash and installing the application again, the behaviour is the same. According to the docs, the default Rendezvous mode for ESP32 is BLE. You can also see in the logs that the BLE advertisement has been started:

```
I (3130) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (3130) chip[DL]: Device already advertising, stop active advertisement and restart
I (3150) NimBLE: GAP procedure initiated: stop advertising.

I (3150) NimBLE: GAP procedure initiated: advertise; 
I (3160) NimBLE: disc_mode=2
I (3160) NimBLE:  adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (3170) NimBLE: 
```

dtodor, 2022-02-16:
I'm building the python controller for Linux, will try to pair the device using it.

dtodor, 2022-02-16:
So, I can confirm that pairing with the Linux python controller (Debian 11) works fine. So this should be Darwin related.

Am I the only one experiencing this error? No one else is using a Mac for development?

dhrishi, 2022-02-16:
>I'm building the python controller for Linux, will try to pair the device using it.
It should work fine with Linux chip-tool as well. 

Also, if it is Monterey then there were some issues related to BLE AFAIK cc @vivien-apple

danh-geo, 2022-03-01:
I updated to Monterey 12.2.1 today and seeing the same issue. It was working yesterday on Big Sur.

Log attached of attempting to pair with standalone CHIP tool
[chip-tool-log.txt](https://github.com/project-chip/connectedhomeip/files/8161272/chip-tool-log.txt)
:

tlykkeberg-grundfos, 2022-03-03:
This is most likely due to iOS (>=15.1) and macOS (>=12.2.1) somehow "swallowing" the Bluetooth Service UUID 0xFFF6 and not letting it onto the application layer. I have tried to change the UUID in both the DUT software and the CHIP stack to something else (i.e. 0x0FF6), which makes it work on iOS and macOS.

tlykkeberg-grundfos, 2022-03-03:
> @tlykkeberg-grundfos which files do I specifically need to update (and where) for me to try this? Thanks

First you need to change the Service UUID which the chip-tool is looking for. This is specified in `src/ble/BleUUID.cpp` in line 31, which holds the `CHIP_BLE_SVC_ID`. I changed it to `0x00, 0x00, 0x0F, 0xF6, 0x00 ....`

Then you would need to change it for your particular platform (mine is CC2652R7 from TI), but the files you would need to change is in the `src/platform/....` directory. For the TI platform it is: `src/platform/cc13x2_26x2/chipOBleProfile.h` in which I changed the `CHIPOBLE_SERV_UUID` from `0xFFF6` to `0x0FF6`.

danh-geo, 2022-03-03:
Thanks for the quick reply @tlykkeberg-grundfos 

I tried your suggestion (although perhaps I did something wrong). 

```
modified:   ../../../src/ble/BleUUID.cpp
```
Modified line 31, which holds the CHIP_BLE_SVC_ID. I changed it to 0x00, 0x00, 0x0F, 0xF6, 0x00

```
modified:   ../../../src/platform/Ameba/BLEManagerImpl.cpp
modified:   ../../../src/platform/CYW30739/cycfg_gatt_db.h
modified:   ../../../src/platform/ESP32/nimble/BLEManagerImpl.cpp
modified:   ../../../src/platform/Linux/BLEManagerImpl.cpp
modified:   ../../../src/platform/Linux/bluez/Helper.cpp
modified:   ../../../src/platform/P6/cycfg_gatt_db.h
modified:   ../../../src/platform/Zephyr/BLEManagerImpl.cpp
modified:   ../../../src/platform/cc13x2_26x2/chipOBleProfile.h
modified:   ../../../src/platform/mbed/BLEManagerImpl.cpp
modified:   ../../../src/platform/nxp/k32w/k32w0/gatt_uuid128.h
```
I'll be honest, I wasn't sure which file to modify, so I modified all the files in the `src/platform/....` directory which mentioned `0xFFF6` to `0x0FF6` 

I then went through the following:
```
cd ${HOME}/tools/esp-idf
. ./export.sh
cd ~/Desktop/connectedhomeip
source ./scripts/bootstrap.sh
source ./scripts/activate.sh
cd ~/Desktop/connectedhomeip/examples/all-clusters-app/esp32
idf.py set-target esp32
idf.py build
idf.py -p /dev/cu.SLAB_USBtoUART erase_flash
idf.py -p /dev/cu.SLAB_USBtoUART flash monitor
```

And then finally attempted pairing again:
```
cd ~/Desktop/connectedhomeip/examples/chip-tool
./out/debug/chip-tool pairing ble-wifi 112233 {ssid} {pw} 20202021 3840
```

But still the pairing is failing on Monterey 12.2.1. As mentioned, it works fine on my laptop which is using Big Sur.

Log attached of the pairing: 
[log.txt](https://github.com/project-chip/connectedhomeip/files/8176599/log.txt)

tlykkeberg-grundfos, 2022-03-03:
@danh-geo It is a different error you see now. But perhaps this is because you have failed to setup the Service UUID (both long and short). You are using the Espressif platform (ESP32), but there are 2 variants, I noticed, in the `src/platform/ESP32` directory.

`src/platform/ESP32/nimble/BLEManagerImpl.cpp`:

```
const ble_uuid16_t ShortUUID_CHIPoBLEService = { BLE_UUID_TYPE_16, 0xFFF6 };
```

This must be changed to `0x0FF6`.

Or perhaps in: `src/platform/ESP32/bluedroid/BLEManagerImpl.cpp`:
```
const uint8_t UUID_CHIPoBLEService[]       = { 0xFB, 0x34, 0x9B, 0x5F, 0x80, 0x00, 0x00, 0x80,
                                         0x00, 0x10, 0x00, 0x00, 0xF6, 0xFF, 0x00, 0x00 };
const uint8_t ShortUUID_CHIPoBLEService[]  = { 0xF6, 0xFF };
```

There you would need to change both the long (128-bit) and the short (16-bit) UUID.

danh-geo, 2022-03-03:
@tlykkeberg-grundfos 

So the first change you mentioned for `src/platform/ESP32/nimble/BLEManagerImpl.cpp` I had already changed and was part of my modified files in my last comment.

I then modified `src/platform/ESP32/bluedroid/BLEManagerImpl.cpp` as you suggested. This is what I changed it too:

```
const uint8_t UUID_CHIPoBLEService[]       = { 0xFB, 0x34, 0x9B, 0x5F, 0x80, 0x00, 0x00, 0x80,
                                         0x00, 0x10, 0x00, 0x00, 0xF6, 0x0F, 0x00, 0x00 };
const uint8_t ShortUUID_CHIPoBLEService[]  = { 0xF6, 0x0F };
```

However the pairing still failed. Appreciating the help thus far.

Log attached.
[log.txt](https://github.com/project-chip/connectedhomeip/files/8176865/log.txt)

dtodor, 2022-03-03:
OK, but now I'm getting the same error also on Linux. I've reverted all the BLE changes, but there seems to be a problem with the all-clusters-app.

dtodor, 2022-03-03:
Maybe problem related to [https://github.com/project-chip/connectedhomeip/commit/907b0f36d3b528250d7efa17ae46e9fd103679e7](https://github.com/project-chip/connectedhomeip/commit/907b0f36d3b528250d7efa17ae46e9fd103679e7)?

s-mcclain, 2022-03-15:
@dtodor have you resolved this problem or still having issues with macOS chip-tool?

dtodor, 2022-03-15:
I think I need to update to macOS 12.3, haven't done this yet.

closed 2022-03-17 by dtodor