project-chip/connectedhomeip #15156: align lock-app/nrfconnect with #14551

msandstedt, 2022-02-14:
#### Problem

#14551 introduces VID / PID check during attestation.  This means that devices must align their advertised VID/PID with their DAC, which in most or all cases will be an example DAC.

Right now, this example has:

```
I: 441 [DL]  Vendor Id: 9050 (0x235A)
I: 445 [DL]  Product Id: 20043 (0x4E4B)
```

 #### Proposed Solution

To match the example DAC, Vendor ID must instead be 0xFFF1 and Product ID must be in the allowable range 0x8000-0x801F.

Damian-Nordic, 2022-02-15:
Which version was this tested on? https://github.com/project-chip/connectedhomeip/commit/482e6fd03196a6de45465a90003947ef4b86e0b1 fixed nrfconnect examples, too and I tested on the ToT lock-app that everything looks fine:
```
I: 454 [DL]  Vendor Id: 65521 (0xFFF1)
I: 458 [DL]  Product Id: 32774 (0x8006)
```

Damian-Nordic, 2022-02-15:
Only lighting-app had a wrong PID set, so fixed that in https://github.com/project-chip/connectedhomeip/pull/15195.

msandstedt, 2022-02-28:
@Damian-Nordic ,

Thanks, yes this is now working.

closed 2022-02-28 by msandstedt