project-chip/connectedhomeip #15174: Build failed when trying to compile on-network (Use full IP-based commissioning = 1)

vinay-apple, 2022-02-15:
#### Problem
Build failed when trying to compile on-network (Use full IP-based commissioning = 1)

SHA: 75c78453086ca0a03895632c3cc3cb4b1c8cd91b

Steps to reproduce

1. On configure idf.py do as follows: 
2. Component config->Bluetooth ->Off
3. Demo->Rendezvous Mode = "On-Network"
4. Component config-> Chip Device Layer -> WiFi Station Options - set up ssid and password
5. Component config-> Chip Device Layer -> Commissioning Options -> Use full IP-based commissioning = 1

Try to compile using command idf.py -p /dev/tty.SLAB_USBtoUART flash monitor and seeing below build failure.

At global scope:
cc1plus: error: unrecognized command line option '-Wno-unknown-warning-option' [-Werror]
cc1plus: all warnings being treated as errors
[314/569] c++ obj/third_party/connectedhomeip/src/protocols/bdx/libBdx.BdxUri.cpp.o
[315/569] c++ obj/third_party/connectedhomeip/src/protocols/echo/libChipProtocols.EchoClient.cpp.o
[316/569] c++ obj/third_party/connectedhomeip/src/platform/ESP32/ESP32.NetworkCommissioningWiFiDriver.cpp.o
[317/569] c++ obj/third_party/connectedhomeip/src/protocols/user_directed_commissioning/libChipProtocols.UserDirectedCommissioningServer.cpp.o
[318/569] c++ obj/third_party/connectedhomeip/src/protocols/echo/libChipProtocols.EchoServer.cpp.o
[319/569] c++ obj/third_party/connectedhomeip/src/protocols/bdx/libBdx.BdxMessages.cpp.o
[320/569] c++ obj/third_party/connectedhomeip/src/protocols/secure_channel/libChipProtocols.MessageCounterManager.cpp.o
[321/569] c++ obj/third_party/connectedhomeip/src/protocols/secure_channel/libSecureChannel.CASEServer.cpp.o
[322/569] c++ obj/third_party/connectedhomeip/src/protocols/user_directed_commissioning/libChipProtocols.UserDirectedCommissioningClient.cpp.o
[323/569] c++ obj/third_party/connectedhomeip/src/protocols/bdx/libBdx.BdxTransferSession.cpp.o
[324/569] c++ obj/third_party/connectedhomeip/src/protocols/secure_channel/libSecureChannel.CASESession.cpp.o
[325/569] c++ obj/third_party/connectedhomeip/src/protocols/bdx/libBdx.TransferFacilitator.cpp.o
[326/569] c++ obj/third_party/connectedhomeip/src/protocols/secure_channel/libSecureChannel.PASESession.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

[Build-Failed-On-Network-Commisioning-M5-Logs.txt](https://github.com/project-chip/connectedhomeip/files/8065205/Build-Failed-On-Network-Commisioning-M5-Logs.txt)

jadhavrohit924, 2022-02-15:
With the exact same steps I am able to compile and perform on-network commissioning and cluster control on the latest codebase (commit: f05070aed57f941533f13b6a1d9b6871b1f930d7)

dhrishi, 2022-02-15:
@vinay-apple Please give it a try and see if it works

cecille, 2022-02-25:
Some of these flags are obsolete. I've put up a PR with updated docs here - https://github.com/project-chip/connectedhomeip/pull/15591. Though I'm not sure that will fix your build errors. Can you build M5 with the default config?

dhrishi, 2022-03-03:
Closing this now. @vinay-apple Feel free to re-open in case it doesn't work.

closed 2022-03-03 by dhrishi