project-chip/connectedhomeip #15182: Removing a fabric does not immediately remove subscriptions for that fabric

bzbarsky-apple, 2022-02-15:
#### Problem
First run:
```
scripts/examples/gn_build_example.sh examples/all-clusters-app/linux out/debug/standalone chip_config_network_layer_ble=false
scripts/examples/gn_build_example.sh examples/chip-tool out/debug/standalone/
```
Then in terminal 1 run:
```
./out/debug/standalone/chip-all-clusters-app
```
In terminal 2 run:
```
./out/debug/standalone/chip-tool pairing onnetwork 17 20202021
./out/debug/standalone/chip-tool administratorcommissioning open-basic-commissioning-window 600 17 0 --timedInteractionTimeoutMs 10000
./out/debug/standalone/chip-tool pairing onnetwork 18 20202021 --commissioner-name beta
./out/debug/standalone/chip-tool onoff subscribe on-off 0 20 1 17 1
```
In terminal 3 run:
```
./out/debug/standalone/chip-tool operationalcredentials remove-fabric 1 18 0 --commissioner-name beta
```

Wait ~10 seconds.  The server logs:
```
[1644901531286] [74728:18584874] CHIP: [DMG] <RE> Sending report (payload has 15 bytes)...
[1644901531286] [74728:18584874] CHIP: [-] ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/ReadHandler.cpp:210: CHIP Error 0x00000003: Incorrect state at ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src
[1644901531286] [74728:18584874] CHIP: [DMG] <RE> Error sending out report data with ../../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/app/ReadHandler.cpp:210: CHIP Error 0x00000003: Incorrect state!
```
which is presumably using a stale fabric index and so forth.

 #### Proposed Solution
Shut down all pending IM handlers for a fabric when it's removed, except maybe the CommandHandler delivering the RemoveFabric command.

@yunhanw-google

erjiaqing, 2022-03-04:
While reading the code, the problem is more complex if we want to shutdown all related im transactions when removing the fabrics.
I believe what need to close for this issue is to shutdown all exchange contexts on that fabric index, they could by any protocol, not only the im protocols.

For example, we should shutdown ongoing transactions like BDX, CASE etc.

I'm going to remove the default implementation of `OnExchangeClosing`, and do the similar thing as `OnResponseTimeout` to the delegates.

Infact, once we finished the above change, this issue should be resolved without other changes required, since the session handle will close all the exchange contexts holding it, and we will clear the SessionHandle when removing the fabric.

Related code:

https://github.com/project-chip/connectedhomeip/blob/e87c25d6c46e2533ff3deb6bdebd9b7d9a6c67a0/src/app/clusters/operational-credentials-server/operational-credentials-server.cpp#L356-L360

https://github.com/project-chip/connectedhomeip/blob/c275d1fc13492a4bf5d7bf3a559af080cb3a75c8/src/transport/SessionManager.cpp#L332-L342

https://github.com/project-chip/connectedhomeip/blob/c11638104320c92ea8b654c889db1c21ed13a1a3/src/transport/Session.h#L88-L95

bzbarsky-apple, 2022-03-05:
> I believe what need to close for this issue is to shutdown all exchange contexts on that fabric index

That already happens, as far as I know.  That's what the `ExpireAllPairingsForFabric` bit is about.

But subscriptions may not have a live exchange, right?  They might be sitting there waiting for something to report.

So sessions and exchanges are a red herring here, for solving _this_ problem.  What we need to do here is get rid of subscriptions.

erjiaqing, 2022-03-05:
> > I believe what need to close for this issue is to shutdown all exchange contexts on that fabric index
> 
> That already happens, as far as I know. That's what the `ExpireAllPairingsForFabric` bit is about. That will notify timeouts on any exchanges that are waiting for a response and associated with any sessions for the fabric.
> 
> But subscriptions may not have a live exchange, right? They might be sitting there waiting for something to report.
> 
> So sessions and exchanges are a red herring here, for solving _this_ problem. What we need to do here is get rid of subscriptions.

Yes, for most async operations that has an active exchange context, ensuring the proper callback is invoked is enough. 

The ReadHandler is holding a SecureSession without implementing the SessionHolder interface (and holding a dangling reference to the exchange context, I will do a clean up and implement SessionHolder for the read handler.

Another issue is the CommandHandler is not implementing the ExchangeDelegate but can also be async, this should also be fixed in the PR for this.

bzbarsky-apple, 2022-03-05:
> The ReadHandler is holding a SecureSession without implementing the SessionHolder interface

That is probably a problem, yes.

> and holding a dangling reference to the exchange context

Dangling where?  For a subscribe it gets nulled out in OnStatusResponse as needed (when we are done with that exchange and it's going to close). Then if we need to send a new report we create a new exchange in `SendStatusReport`.

(I believe you that it might be dangling somewhere; I just haven't found where yet.)

> Another issue is the CommandHandler is not implementing the ExchangeDelegate but can also be async

Yes, that is a bug, but I don't think it affects this issue per se...

PR #15901 (2022-03-08): [IM] Abort read/subscribe transactions when deleting related fabric
#### Problem
Fixes #15182 

#### Change overview
Adds `CloseTransactionsFromFabricIndex` for shutting down a ReadHandler when deleting related fabric

#### Testing

Send the following command via REPL:
```
In [1]: devCtrl.CommissionIP(b"172.16.243.197", 20202021, 233)

In [2]: res = await devCtrl.ReadAttribute(
   ...:     nodeid=233, attributes=[("*")], reportInterval=(10, 20), returnClusterObject=True
   ...: )

In [5]: cmd = await devCtrl.SendCommand(
   ...:     nodeid=233,
   ...:     endpoint=0,
   ...:     payload=Clust [... cut]
closed 2022-03-08 by woody-apple