project-chip/connectedhomeip #15191: How to config force use Python3 for all builds?

TE-N-ElvenWang, 2022-02-15:
Hi,
I had a build environment integration with some other conponents. The build environment has Python2 and Python3. But the default /usr/bin/python points to python2 due some reasons.
Also virtualenv is not available in my environment. So I'd like to check whether there is some way to config the connectedhomeip builds use "python3" instead of "python"? 
As I'm facing below error when build all-cluster examples:

```
ninja: Entering directory `out/all-cluster'
[450/514] ACTION //third_party/connectedhomeip/third_party/pigweed/repo/pw_build:python_lint(//third_party/connectedhomeip/build/toolchain/custom:custom)
FAILED: gen/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/python_lint.pw_pystamp
python ../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../examples/all-clusters-app/linux/ --current-path ../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_build --default-toolchain=//third_party/connectedhomeip/build/toolchain/custom:custom --current-toolchain=//third_party/connectedhomeip/build/toolchain/custom:custom --touch gen/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/python_lint.pw_pystamp --capture-output -- ../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/nop.py
  File "../../examples/all-clusters-app/linux/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py", line 36
    def _parse_args() -> argparse.Namespace:
                      ^
```

Thanks!

andy31415, 2022-02-16:
Python3 seems like a reasonable minimal requirement to build CHIP: python2 sunset date is Jan 1 2020: https://www.python.org/doc/sunset-python-2/#:~:text=We%20have%20decided%20that%20January,as%20soon%20as%20you%20can.

I would prefer to not make any sort of effort towards trying to  support python2 as that takes development resources.

mspang, 2022-02-19:
Set script_executable to python3 in your dotfile (.gn)

https://gn.googlesource.com/gn/+/master/docs/reference.md#other-help-topics-gn-file-variables

closed 2022-02-19 by mspang