project-chip/connectedhomeip #15223: Pairing one of of the M5 boards always failed with Failed during PASE session setup. Error CHIP:0x00000038

vinay-apple, 2022-02-15:
#### Problem

Pairing one of of the M5 boards always failed with Failed during PASE session setup. Error CHIP:0x00000038

SHA: 17dc4b003f2a433d2220cb3c22829d6336360e02

Steps to reproduce

1. Pair M5 board with chip tool on iOS and see that it fails on below error 
2. Tried 5 times resetting M5 board and relaunching chip tool but still hit below error all 5 times. 

(35316) chip[EM]: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:2151800833 on exchange 24107r
E (35326) chip[SC]: Received error (protocol code 2) during PASE process. Error CHIP:0x00000038
E (35336) chip[SC]: Failed during PASE session setup. Error CHIP:0x00000038
E (35346) chip[SVR]: Commissioning failed (attempt 1): Error CHIP:0x00000038
I (35356) chip[BLE]: Releasing end point's BLE connection back to application.

Was able to get pairing to work on another M5 board. 

[Failed-during-PASE-session-setup-chiptool-logs.txt](https://github.com/project-chip/connectedhomeip/files/8075185/Failed-during-PASE-session-setup-chiptool-logs.txt)
[Failed-during-PASE-session-setup-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8075186/Failed-during-PASE-session-setup-M5-logs.txt)

woody-apple, 2022-06-14:
@vinay-apple Is this still happening?

woody-apple, 2022-11-02:
Issue Scrub: Given the age of this, closing, please re-open a new issue if needed.

closed 2022-11-02 by woody-apple