project-chip/connectedhomeip #15251: [TC-MC-10.3] & [TC-MC-10.5]-TV-App DUT should support Content Launcher Cluster commands

Rajashreekalmane, 2022-02-16:
#### Problem
While Executing the media cluster test case [TC-MC-10.3] and [TC-MC-10.5]  TV-app DUT is returning command not supported.
TV-app should be enhanced to support the Content launcher commands as mentioned in the Spec.

 * expected behavior:-The DUT should respond with a success response (value 0) 
 * actual behavior:- We are getting UNSUPPORTED_CLUSTER(0xC3) error response.
 
Command used in chip-controller:
1. sudo ./chip-tool contentlauncher launch-content '{"parameterList": [ { "type": 0, "value": "exampleString", "externalIDList": [ { "name": "exampleName", "value": "exampleValue"} ] } ] }' 0 1 0
2. sudo ./chip-tool contentlauncher launch-url "exampleUrl"   '{"providerName":  "exampleName", "background": { "imageUrl": "exampleUrl", "color": "blue", "size" : {"width": 1.0, "height": 1.0,  "metric": 0} }}' 1 0

 Reference screenshot:
[image]

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/media/ContentLauncher.adoc

Test-Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#content-launcher-cluster-test-cases

PFA Logs below:-
[[TC-MC-10.3] & [TC-MC-10.5].txt](https://github.com/project-chip/connectedhomeip/files/8080147/TC-MC-10.3.TC-MC-10.5.txt)

Rajashreekalmane, 2022-03-24:
Above mentioned same issue facing while  verifing that the DUT can properly respond to attribute read requests Content Launcher Cluster Attributes.

Command used in chip-controller:
./chip-tool contentlauncher read accept-header 1 1

PFA DUT screenshot below:-
[image]

Rajashreekalmane, 2022-04-07:
Same issue facing in Application Launcher Cluster.

Command used in chip-controller:
 ./chip-tool applicationlauncher launch-app '{"catalogVendorId": 123, "applicationId": "exampleid"}' 1 1

PFA DUT screenshot below:-
[image]

lazarkov, 2022-04-11:
Hi @Rajashreekalmane , I'm using chip-tool and the first comamnds were not good, they written be like this:
`out/debug/standalone/chip-tool contentlauncher launch-url "exampleUrl"   '{"providerName":  "exampleName", "background": { "imageUrl": "exampleUrl", "color": "blue", "size" : {"width": 1.0, "height": 1.0,  "metric": 0} }}' 1 1`

`out/debug/standalone/chip-tool contentlauncher launch-content '{"parameterList": [ { "type": 0, "value": "exampleString", "externalIDList": [ { "name": "exampleName", "value": "exampleValue"} ] } ] }' 0 1 1 `

These ones work for me:
`./chip-tool applicationlauncher launch-app '{"catalogVendorId": 123, "applicationId": "exampleid"}' 1 1`
`./chip-tool applicationlauncher read catalog-list 1 1`

Can you please check if the suggested commands work for you?

Rajashreekalmane, 2022-04-12:
@lazarkov I am facing same issue.
In controller side I got response but in DUT it's showing "Endpoint 1 not found"
[image]

Command used in chip-controller:
./chip-tool applicationlauncher launch-app '{"catalogVendorId": 123, "applicationId": "exampleid"}' 1 1
./chip-tool applicationlauncher read catalog-list 1 1

closed 2022-04-21 by lazarkov