project-chip/connectedhomeip #15261: ble-wifi command are failing

vijs, 2022-02-16:
#### Problem

https://github.com/project-chip/connectedhomeip/pull/14751 had disabled Wi-Fi and replaced it with Thread.
By making the Thread (as default), the chip-tool pairing ble-wifi command does not work anymore.

 * expected behavior
    chip-tool pairing ble-wifi command should work
 * actual behavior
    chip-tool pairing ble-wifi command is not working in the latest master
 * steps to reproduce
    Run "chip-tool pairing ble-wifi" command

tcarmelveilleux, 2022-02-16:
Please provide more details about:

- What is the config at commissioning side
- What is the config at commissionee side
- Logs for both sides

tcarmelveilleux, 2022-02-16:
Slack resolution: lighting-app was being used, which does not have proper Network Commissioning cluster setup on LInux.

closed 2022-02-16 by tcarmelveilleux