project-chip/connectedhomeip #15278: Enable OTA Provider server for controllers

carol-apple, 2022-02-16:
#### Problem
Currently, to test OTA updates, the ota-provider-app and ota-requestor-app are used. In a more realistic scenario, a controller would likely be the OTA Provider. The OTA Provider server should be enabled for controllers

 #### Proposed Solution
Enable OTA Provider server in `src/controller/data_model/controller-clusters.zap`

bzbarsky-apple, 2022-03-08:
Controllers will need to do this as they choose, depending on their endpoint configuration.

For Darwin, this is now set up, with #15932 and https://github.com/project-chip/connectedhomeip/pull/15887

carol-apple, 2022-03-08:
Updating this issue to track the changes for Darwin only

bzbarsky-apple, 2022-03-08:
In that case, I think this is more or less fixed.  Remaining bits:

1. Make the Darwin server do mdns advertising.  Created https://github.com/project-chip/connectedhomeip/pull/15951 for that.
2. Ensure that other nodes can create CASE connections to the Darwin server.  Still sorting that out.
3. Implement an OTA provider delegate, etc.

carol-apple, 2022-03-23:
The following PRs address the bare minimum needed to support OTA provider for Darwin controllers:

- https://github.com/project-chip/connectedhomeip/pull/15885
- https://github.com/project-chip/connectedhomeip/pull/15887
- https://github.com/project-chip/connectedhomeip/pull/15932
- https://github.com/project-chip/connectedhomeip/pull/16504
- https://github.com/project-chip/connectedhomeip/pull/16546

closed 2022-03-23 by carol-apple