project-chip/connectedhomeip #15283: ESP32/M5 board: Reporting stops after M5 board crashes

kean-apple, 2022-02-16:
#### Problem
Reporting on ESP32/M5 board stops when the M5 board crashes

1. Pair M5 board with iOS chiptool
2. Go to Temp sensor, enable reporting min: 1s, max: 2s, delta: 1C
3. Verify that M5 board is sending reports every 1s and iOS chiptool is shows updated temp sensor reading when changed on the M5 board
4. Let M5 board sit for a while..M5 board crashed (will file a separate GitHub for crash)
5. After crash, M5 board doesn't resume reporting. Modify temp sensor on M5 board..its not being reflected on iOS chip tool

In Step 5, I would expect that reporting should resume after recovering from crash. Use case in the field, if an accessory does crash it should resume reporting otherwise the App that's consuming reports will not get updated accessory state changes eg. temp sensor increasing temp 
[M5-board-stops-reporting-after-reset.txt](https://github.com/project-chip/connectedhomeip/files/8084210/M5-board-stops-reporting-after-reset.txt)

kean-apple, 2022-02-16:
Filed separate issue for the crash: https://github.com/project-chip/connectedhomeip/issues/15285

bzbarsky-apple, 2022-05-21:
This is expected behavior, right?  Subscriptions are not permanent...

closed 2022-05-21 by bzbarsky-apple