project-chip/connectedhomeip #15285: M5 board crashed after a couple of minutes with reporting enabled -  abort() was called at PC 0x40125ed1 on core 1

kean-apple, 2022-02-16:
#### Problem
M5 board crashed after a couple of minutes with reporting enabled

SHA: f7910c98ed2ebd00cea4b095c8cce220d32aa238

1. Pair M5 board with iOS chip tool app
2. Go to temp sensor to enable reporting with min: 1s, max:2s , Delta: 1C
3. After a couple of minutes the M5 board crashes

CRASH:
abort() was called at PC 0x40125ed1 on core 1
0x40125ed1: chip::Inet::UDPEndPointImplLwIP::LwIPReceiveUDPMessage(void*, udp_pcb*, pbuf*, ip_addr const*, unsigned short) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:477
 (inlined by) chip::ReferenceCounted<chip::Inet::UDPEndPoint, chip::Inet::EndPointDeletor<chip::Inet::UDPEndPoint>, 1>::Retain() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/core/ReferenceCounted.h:55
 (inlined by) chip::Inet::UDPEndPointImplLwIP::LwIPReceiveUDPMessage(void*, udp_pcb*, pbuf*, ip_addr const*, unsigned short) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/inet/UDPEndPointImplLwIP.cpp:374

 (2755896) chip[EM]: Sending Standalone Ack for MessageCounter:779412 on exchange 59451i
E (2755906) chip[IN]: Prepared secure message 0x3ffe8504 to 0x1725AE4EA7E0E4CD (1)  of type 0x10 and protocolId (0, 0) on exchange 59451i with MessageCounter:2400309.
E (2755926) chip[IN]: Sending encrypted msg 0x3ffe8504 with MessageCounter:2400309 to 0x1725AE4EA7E0E4CD (1) at monotonic time: 2755139 msec
E (2755936) chip[EM]: Flushed pending ack for MessageCounter:779412 on exchange 59451i
E (2755946) chip[DL]: Long dispatch time: 124 ms, for event type 3
E (2756166) chip[DMG]: Unblock report hold after min 1 seconds

abort() was called at PC 0x40125ed1 on core 1
Backtrace:0x40081e16:0x3ffdc3200x40092281:0x3ffdc340 0x40098916:0x3ffdc360 0x40125ed1:0x3ffdc3d0 0x4016f81d:0x3ffdc410 0x401725b9:0x3ffdc460 0x40176082:0x3ffdc490 0x40169629:0x3ffdc4b0 0x40095485:0x3ffdc4e0 
0x40081e16: panic_abort at /Users/keanhinlim/tools/esp-idf/components/esp_system/panic.c:402
0x40092281: esp_system_abort at /Users/keanhinlim/tools/esp-idf/components/esp_system/esp_system.c:121
0x40098916: abort at /Users/keanhinlim/tools/esp-idf/components/newlib/abort.c:46
0x40125ed1: chip::Inet::UDPEndPointImplLwIP::LwIPReceiveUDPMessage(void*, udp_pcb*, pbuf*, ip_addr const*, unsigned short) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:477
 (inlined by) chip::ReferenceCounted<chip::Inet::UDPEndPoint, chip::Inet::EndPointDeletor<chip::Inet::UDPEndPoint>, 1>::Retain() at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/core/ReferenceCounted.h:55
 (inlined by) chip::Inet::UDPEndPointImplLwIP::LwIPReceiveUDPMessage(void*, udp_pcb*, pbuf*, ip_addr const*, unsigned short) at /Users/keanhinlim/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/inet/UDPEndPointImplLwIP.cpp:374
0x4016f81d: udp_input at /Users/keanhinlim/tools/esp-idf/components/lwip/lwip/src/core/udp.c:404
0x401725b9: ip4_input at /Users/keanhinlim/tools/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:778
0x40176082: ethernet_input at /Users/keanhinlim/tools/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
0x40169629: tcpip_thread_handle_msg at /Users/keanhinlim/tools/esp-idf/components/lwip/lwip/src/api/tcpip.c:180
 (inlined by) tcpip_thread at /Users/keanhinlim/tools/esp-idf/components/lwip/lwip/src/api/tcpip.c:154
0x40095485: vPortTaskWrapper at /Users/keanhinlim/tools/esp-idf/components/freertos/port/xtensa/port.c:130

kean-apple, 2022-02-16:
[M5-board-stops-reporting-after-reset.txt](https://github.com/project-chip/connectedhomeip/files/8084308/M5-board-stops-reporting-after-reset.txt)

woody-apple, 2022-06-14:
(Please close if not!)

dhrishi, 2022-07-14:
Thanks @shubhamdp . Closing this for now. @kean-apple Feel free to reopen if you face the issue

closed 2022-07-14 by dhrishi