project-chip/connectedhomeip #15288: [TC - LC -1.5] User Label cluster empty entry verification - Not able to write

cabin15, 2022-02-17:
#### Problem
Tested on master branch on commit 5cb8800a7f46bd7e9f62d32499c4aa053b0884fd.
Used Tool: 
- chip-tool
- chip-all-clusters-app

I tried to execute step 1 of [TC-LC-1.5], but I'm not able to come up with proper chip-tool command to achieve this.

My test steps.
- Commissioning is already done.
- This is the command I can think of to achieve step 1 of TC-LC-1.5
./chip-tool userlabel write label-list '[{"room":"bedroom 1"}]' 12345678 1

And I got below error logs.
root@raspberrypi4:/home/root# ./chip-tool_210215 userlabel write label-list '[{"room":"Bad"}]' 12345678 1
Usage:
  ./chip-tool_210215 userlabel write attr-name attr-value node-id endpoint-id [--commissioner-name] [--ble-adapter] [--timedInteractionTimeoutMs] [--data-version]
Feb 16 22:24:38 raspberrypi4 chip-tool_210215[4593]: LabelStruct.label is required.  Should be provided as {"label": value}
Feb 16 22:24:38 raspberrypi4 chip-tool_210215[4593]: Run command failure: ../examples/chip-tool/commands/common/Commands.cpp:135: Error 0x0000002F
root@raspberrypi4:/home/root#

I dig up a little bit more, and found a below pull request.
https://github.com/project-chip/connectedhomeip/pull/13761

It seems like above PR supports the complex type as an argument.

@vivien-apple ,
Could you plz help me with this?
TC says, "TH1 writes LabelList attribute of the DUT by setting "room:bedroom 1","Orientation:east", "", "", "", "direction:down"
Could you plz let me know what chip-tool command could achieve this?
I appreciate your support.

cabin15, 2022-02-17:
@vivien-apple 
Could you plz help me with this to come up with proper command to achieve TC-LC-1.5?
Thank you in advance.

markus-becker-tridonic-com, 2022-03-11:
This issue also affects [TC-IDM-3.2] Step 10.

andy31415, 2022-05-30:
This should be fixed for linux, nrf and efr. Other platforms will need to implement DeviceInfoProbider interface (or likely just use the example one to start).

closed 2022-05-30 by andy31415