project-chip/connectedhomeip #15296: TV_AccountLoginCluster.yaml failing in Linux platform

manjunathambiger23, 2022-02-17:
Summary:
               TV_AccountLoginCluster.yaml testcase seems to be failing when we execute in the Linux system

Execution log and Screenshot:
[image]
[Uploading faillog.zip…]()

manjunathambiger23, 2022-02-18:
> @manjunathambiger23 Are you executing it against tv-app, or all-clusters-app?

@bzbarsky-apple we are executing it against all-clusters-app ,when we try to use this command ./scripts/tests/run_test_suite.py --log-level debug run we are getting this error

bzbarsky-apple, 2022-02-18:
The TV cluster tests need to be run against the tv app.  See how CI does it.  all-clusters-app just does not implement the relevant clusters in a meaningful way.

closed 2022-02-22 by manjunathambiger23