project-chip/connectedhomeip #15299: Issue in commissioning device via iOS CHIPTool app

ajay-gantayet, 2022-02-17:
#### Problem

I am unable to commission matter device using the CHIPTool app. 
* Expected behaviour: I should be able to commission all-cluster-app/esp32 matter device with the given CHIPTool app. 
* Actual behavior: After the call to [- (BOOL)commissionDevice:(uint64_t)deviceId commissioningParams (CHIPCommissioningParameters *)commissioningParams error:(NSError * __autoreleasing *)error;] we are getting a failure.
* Steps to reproduce:
   - Build and run the CHIPTool app
   - Run the firmware on esp32c3 board
   - Try and commission the device using the CHIPTool app.
* System configuration:
   - The commit ID I am using is: b03bc9fb057170ff958b790c0d08b473c090468e
* PFA logs for the above:
[device_logs.txt](https://github.com/project-chip/connectedhomeip/files/8088664/device_logs.txt)
[iOS_app_logs.txt](https://github.com/project-chip/connectedhomeip/files/8088665/iOS_app_logs.txt)
* Note:
   - When I changed the **chip_mdns** value from **platform** to **minimal** for **chip_device_platform == "darwin"**, the commissioning worked.

We should be able to commissiond evice with the default settings. Could someone please hlep me with this?

dhrishi, 2022-02-17:
@woody-apple @bzbarsky-apple Can you please check and assign it to the relevant folks

bzbarsky-apple, 2022-02-17:
@sagar-apple any idea what's going on here?

What I see in the logs is that we complete the SPAKE2 handshake, and then:
```
2022-02-17 17:29:09.746507+0530 CHIPTool[3225:2072935] [all] 🔴 [1645099149746] [3225:2072935] CHIP: [CTL] Invalid device for commissioning 0000000000000000
2022-02-17 17:29:09.746552+0530 CHIPTool[3225:2072935] Error(../../../../../../../../../../../Documents/Projects/Matter/connectedhomeip/src/controller/CHIPDeviceController.cpp:885: CHIP Error 0x00000003: Incorrect state): Failure while pairing the device
2022-02-17 17:29:09.746734+0530 CHIPTool[3225:2072935] Failed to commission Device 0, with error Error Domain=CHIPErrorDomain Code=6 "Invalid object state." UserInfo={NSLocalizedDescription=Invalid object state.}
```

Those first two log lines are from this code in `DeviceCommissioner::Commission`:
```
    CommissioneeDeviceProxy * device = mDeviceBeingCommissioned;
    if (device == nullptr || device->GetDeviceId() != remoteDeviceId ||
        (!device->IsSecureConnected() && !device->IsSessionSetupInProgress()))
    {
        ChipLogError(Controller, "Invalid device for commissioning " ChipLogFormatX64, ChipLogValueX64(remoteDeviceId));
        return CHIP_ERROR_INCORRECT_STATE;
    }
```

But what I don't get is what chip_mdns has to do with any of this: we are failing out well before we have done any sort of mdns resolution apart from maybe on-network commissionable discovery... but it sure looks like we are discovering the right device in terms of talking to it, so I would not expect that to affect things here.

sagar-apple, 2022-02-17:
Just prior to that, I wonder if this had anything to do with it. 

```
2022-02-17 17:28:52.876959+0530 CHIPTool[3225:2072935] [all] 🔵 [1645099132876] [3225:2072935] CHIP: [BLE] NewConnection
2022-02-17 17:28:52.881801+0530 CHIPTool[3225:2072935] [all] 🔴 [1645099132881] [3225:2072935] CHIP: [DL] Browse (kDNSServiceErr_NoAuth)
2022-02-17 17:28:52.881921+0530 CHIPTool[3225:2072935] Error(../../../../../../../../../../../Documents/Projects/Matter/connectedhomeip/src/platform/Darwin/DnssdImpl.cpp:390: CHIP Error 0x000000AC: Internal error): Failure while pairing the device
```

For some reason looks like the app didn't have mDNS permission(User consent?) and that caused the mDNS pairing flow to early return and possibly clean up state internally. At the same time BLE Spake2+ start going through and once it was done, we were in a screwed up state.

sagar-apple, 2022-02-17:
Ajay if you can repro this, it would be good to know if you explicitly allow CHIPTool local network access (Settings->CHIPTool->Local Network) and try this again. 

Without changing mDNS.

ajay-gantayet, 2022-02-19:
Hi @sagar-apple  I tested the scenario with default chip_mdns. I made sure that the app local network permission.
Still not able to commission device successfully. 

On the app side, the 
```
- (BOOL)commissionDevice:(uint64_t) deviceId  commissioningParams:(CHIPCommissioningParameters *) commissioningParams  error:(NSError * __autoreleasing *) error
```
method is giving the following error:
```
**2022-02-19 13:18:19.647503+0530 CHIPTool[3840:2409079] [all] 🔴 [1645256899647] [3840:2409079] CHIP: [CTL] Invalid device for commissioning 0000000000000000
2022-02-19 13:18:19.647542+0530 CHIPTool[3840:2409079] Error(../../../../../../../../../../../Documents/Projects/Matter/new/connectedhomeip/src/controller/CHIPDeviceController.cpp:885: CHIP Error 0x00000003: Incorrect state): Failure while pairing the device
2022-02-19 13:18:19.647661+0530 CHIPTool[3840:2409079] Failed to commission Device 0, with error Error Domain=CHIPErrorDomain Code=6 "Invalid object state." UserInfo={NSLocalizedDescription=Invalid object state.}**

```
[esp32_device_logs.txt](https://github.com/project-chip/connectedhomeip/files/8101659/esp32_device_logs.txt)
[ios_app_logs.txt](https://github.com/project-chip/connectedhomeip/files/8101660/ios_app_logs.txt)
[image]

sagar-apple, 2022-02-19:
Hey Ajay! Thanks for checking again. So weird, I see the same dns error that's causing an early return and terminating the pairing. 

```
2022-02-19 13:18:11.958620+0530 CHIPTool[3840:2409079] [all] 🔴 [1645256891958] [3840:2409079] CHIP: [DL] Browse (kDNSServiceErr_NoAuth)
2022-02-19 13:18:11.958736+0530 CHIPTool[3840:2409079] Error(../../../../../../../../../../../Documents/Projects/Matter/new/connectedhomeip/src/platform/Darwin/DnssdImpl.cpp:390: CHIP Error 0x000000AC: Internal error): Failure while pairing the device
```

Can you make sure the you have the following items in CHIPTool's info.plist (they should be there by default). If they're somehow not then this would fail.

```
	<key>NSBonjourServices</key>
	<array>
		<string>_matter._tcp</string>
		<string>_matterc._udp</string>
		<string>_matterd._udp</string>
	</array>
```

If you have those, it might be worth printing the `regtype` inside `src/platform/Darwin/DnssdImpl.cpp` 
print the c_str being passed into the iOS DNSServiceBrowse command. That way we can try to repro it locally by isolating to just what's being asked for here.

```
    err   = DNSServiceBrowse(&sdRef, 0 /* flags */, interfaceId, regtype.c_str(), kLocalDot, OnBrowse, sdCtx);
```

sagar-apple, 2022-02-22:
@ajay-gantayet What iOS version are you on? Can you try the latest beta builds?

ajay-gantayet, 2022-02-23:
> @ajay-gantayet What iOS version are you on? Can you try the latest beta builds?

@sagar-apple  my iOS version is 15.1.

sagar-apple, 2022-02-23:
> > @ajay-gantayet What iOS version are you on? Can you try the latest beta builds?
> 
> 
> 
> @sagar-apple  my iOS version is 15.1.

Are you able to try the latest iOS 15.4 beta?

ajay-gantayet, 2022-02-23:
I will try it out and update you.

ajay-gantayet, 2022-02-25:
@sagar-apple  I was able to successfully commission device using iOS 15.4 beta.
Thank you for your help!

closed 2022-02-25 by bzbarsky-apple