project-chip/connectedhomeip #15304: Reading Fabric list on M5 board from iOS chip tool occasionally gives error -  Got back fabrics list: (null) error Error Domain=CHIPErrorDomain Code=1 "Undefined error:50."

kean-apple, 2022-02-17:
#### Problem
Reading Fabric list from iOS chip tool occasionally gives error

Steps:
Initial: Load 2 iOS chip tool on same device (different bundle ids)
1. Pair M5 board with Chip Tool 1 - verify toggle light works
2. On Chip tool 1, enable pairing mode 
3. On Chip tool 2, pair M5 board by entering pin code provided by Chip tool 1 - verify toggle light works
4. On both Chip tool 1 & 2, go to Fabric Management and select Update Fabric list - verify it list correct Fabric ID for that app

In step 4, occasionally see error: 
2022-02-17 07:04:32.097476-0800  localhost CHIPTool[9119]: Got back fabrics list: (null) error Error Domain=CHIPErrorDomain Code=1 "Undefined error:50." UserInfo={NSLocalizedDescription=Undefined error:50., errorCode=50}

 This is probably seen 20% of time
[error-returning-fabric-list-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8089789/error-returning-fabric-list-M5-logs.txt)
[error-returning-fabric-list-chiptool-log.txt](https://github.com/project-chip/connectedhomeip/files/8089790/error-returning-fabric-list-chiptool-log.txt)
s

bzbarsky-apple, 2022-02-17:
> "Undefined error:50."

`50` would be `0x00000032` which is a core SDK error with code 0x32.  Looks like this one:
```
#define CHIP_ERROR_TIMEOUT                                     CHIP_CORE_ERROR(0x32)
```

bzbarsky-apple, 2022-02-17:
From the m5stack log (which seems to be cut off toward the beginning):
```
E (574366) chip[IM]: Received Read request
E (574366) chip[IM]: no resource for Read interaction
```

So what I am seeing in the m5stack log (and thank you _very_ much for getting the detail logging going so I can see the payloads!) is, in order

1. Read request for Fabrics.
2. Read request for CurrentFabricIndex.
3. Read request for ... something, but "no resource for Read interaction", send status reply.
4. Send responses for read requests from steps 1, 2.
5. Repeat of steps 1-4.
6. Repeat of steps 1-4.
7. Repeat of steps 1-4.
8. Receive UpdateFabricLabel command with label "Test".
9. Send InvokeResponse.
10. Send Report Data message (responding to what?) for CommisionedFabrics attribute.
11. Read request for Fabrics.
12. Read request for CurrentFabricIndex.
13. Read request for CommissionedFabrics.
14. Send Report Data for steps 11 and 12 (but not 13!)
16. Read request for Fabrics.
17. Read request for CurrentFabricIndex.
18. Read request for ... something, but "no resource for Read interaction", send status reply.
19. Send Report Data for steps 16 and 17
20. Read request for Fabrics.
21. Read request for CurrentFabricIndex.
22. Read request for ... something, but "no resource for Read interaction", send status reply.
23. Send Report Data for steps 20 and 21
24. Receive UpdateFabricLabel command with label "Test 2".
25. Send InvokeResponse.
26. Send Report Data message for CommisionedFabrics attribute (from step 13?).
27. Read request for Fabrics.
28. Read request for CurrentFabricIndex.
29. Read request for CommissionedFabrics.
30. Send Report Data for steps 27 and 28 (but not 29!).
31. Read request for Fabrics.
32. Read request for CurrentFabricIndex.
33. Read request for ... something, but "no resource for Read interaction", send status reply.
34. Send Report Data for steps 31 and 32
35. Get failures after several MRP retransmits.

closed 2022-02-18 by yunhanw-google