project-chip/connectedhomeip #15325: EFR32 lock and lighting app don't boot.

rgoliver, 2022-02-17:
#### Problem
After the additional lock checks from #15152, EFR32 dies at startup (tested on lighting and locking apps):
```
20220217 15:16:36  INF  <efr32 > ==================================================
20220217 15:16:36  INF  <efr32 > chip-efr32-lock-example starting
20220217 15:16:36  INF  <efr32 > ==================================================
20220217 15:16:36  INF  <efr32 > Init CHIP Stack
20220217 15:16:37  INF  <info  > [DL] Setting device name to : "EFR32_LOCK"
20220217 15:16:37  INF  <efr32 > Initializing OpenThread stack
20220217 15:16:37  INF  <info  > [DL] OpenThread started: OK
20220217 15:16:37  INF  <info  > [DL] Setting OpenThread device type to ROUTER
20220217 15:16:37  INF  <efr32 > Starting Platform Manager Event Loop
20220217 15:16:37  INF  <efr32 > Starting OpenThread task
20220217 15:16:37  INF  <efr32 > Starting App Task
20220217 15:16:37  INF  <efr32 > Starting FreeRTOS scheduler
20220217 15:16:37  INF  <info  > [DL] Bluetooth stack booted: v3.2.1-b216
20220217 15:16:37  INF  <info  > [DL] RAIL version:, v2.11.1-b0
20220217 15:16:37  INF  <detail> [DL] Thread task running
20220217 15:16:37  INF  <detail> [-] Starting pw_rpc server
20220217 15:16:37  INF  <detail> [DL] CHIP task running
20220217 15:16:37  INF  <detail> [DL] Start BLE advertissement
20220217 15:16:37  INF  <info  > [ZCL] Using ZAP configuration...
20220217 15:16:37  INF  <error > [DL] Chip stack locking error at '../../examples/lock-app/efr32/third_party/connectedhomeip/src/app/util/attribute-stor
20220217 15:16:37  INF  age.cpp:553'. Code is unsafe/racy
20220217 15:16:37  INF  <error > [-] chipDie chipDie chipDie
```
 #### Proposed Solution
Need to go through startup code and add appropriate locks.

jmeg-sfy, 2022-02-18:
> #### Problem
> After the additional lock checks from #15152, EFR32 dies at startup (tested on lighting and locking apps):
>
> ```
> 20220217 15:16:36  INF  <efr32 > [...]

Problem is likely on every single EXAMPLE => window-covering as well

mkardous-silabs, 2022-02-18:
PR #15365 fixes all the examples for efr32.
This PR, #15152  had added an assert when interacting with the matter stack.

PR #15365 (2022-02-22): Add lock/unlock calls for efr32 examples
#### Problem
* Applications was not holding lock on the matter task when reading/writing or calling commands
* Fixes #15325 

#### Change overview
* Add lock/unlock or scheduleWork to application calls interacting with matter stack

#### Testing
* Manual tests with every examples to verify that app doesn't try to access the matter stack without holding the semaphore
closed 2022-02-22 by jmartinez-silabs