project-chip/connectedhomeip #15341: Cirque Test: For network commissioning test, we need to check if device support network commissioning or not via descriptor cluster

yunhanw-google, 2022-02-18:
#### Problem
Cirque Test: For network commissioning test, we need to check if device support network commissioning or not via descriptor cluster

cecille, 2022-02-25:
Auto commissioner now checks the network cluster to determine what kind of network is supported (thread/wifi) and will error the commissioner is using BLE commissioning but has not supplied the correct parameters. This should already be in place for cirque tests, as they are using this commissioner. So I think this is done. @yunhanw-google - can you confirm that I understand the issue correctly here and close if appropriate?

mrjerryjohns, 2022-03-03:
@erjiaqing can you comment?

erjiaqing, 2022-03-14:
Yes, however, seems the code for setting network commissioning params for the auto commissioner is commented out. I will do some cleanup for the python testing script and re-enable it.

closed 2022-10-08 by stale[bot]