project-chip/connectedhomeip #15348: [TC-ETHDIAG-2.1] PacketTxCount and PacketRxCount attribute value mismatch

manjunathambiger23, 2022-02-18:
Summary:
              As per testplan PacketTxCount and PacketRxCount attribute value must be 0 after sending ResetCounts command but when we execute its failing showing value mismatch for both attribute.

Testplan screenshot:
[image]

Execution logs and screenshot:
[packetRxcount.zip](https://github.com/project-chip/connectedhomeip/files/8097028/packetRxcount.zip)
[image]
[image]
[packetTxCount.zip](https://github.com/project-chip/connectedhomeip/files/8097029/packetTxCount.zip)

bzbarsky-apple, 2022-03-12:
The test plan is wrong.

When ResetCounts is received, the counts are set to 0, sure.  Then a response is sent, which is a transmitted packet, so the TX count becomes 1.  That's the value that is sent in the response.

Similarly, after ResetCount a response is sent and an MRP ack is received, so the RX count becomes 1.  Now a read is sent, which is received, the the RX count becomes 2.  That's the value that is sent in the response.

bzbarsky-apple, 2022-03-12:
Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/941

closed 2022-03-12 by bzbarsky-apple