project-chip/connectedhomeip #15362: Pairing with second instance of chiptool results in failure CHIP Error 0x00000038: Invalid PASE parameter

vinay-apple, 2022-02-18:
#### Problem

Pairing with second instance of chiptool results in failure CHIP Error 0x00000038: Invalid PASE parameter

SHA: 149c271f497b4493dbc009aed8f616922fc9b5fe

Steps to reproduce

1. Pair M5 with chip tool1 and see that toggle light works fine. 
2. Enable pairing from chip tool1 
3. Try to pair M5 with second instance, chip tool 2 (on a different device) and seeing below error.

Note: Same error was seen on a different setup when tried to install chip tool 2 on the same device and tried to pair M5 with it. 

2022-02-18 12:52:01.170431-0800  localhost CHIPTool[4853]: (CHIP) [com.zigbee.chip:all] ð´ [1645217521170] [4853:261377] CHIP: [SC] Received error (protocol code 2) during PASE process. ../../../../../../../../../../../CHIP-TOOL/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:846: CHIP Error 0x00000038: Invalid PASE parameter
2022-02-18 12:52:01.170503-0800  localhost CHIPTool[4853]: (CHIP) [com.zigbee.chip:all] ð´ [1645217521170] [4853:261377] CHIP: [SC] Failed during PASE session setup. ../../../../../../../../../../../CHIP-TOOL/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:846: CHIP Error 0x00000038: Invalid PASE parameter
2022-02-18 12:52:01.170584-0800  localhost CHIPTool[4853]: (CHIP) DevicePairingDelegate status updated: 1
2022-02-18 12:52:01.170791-0800  localhost CHIPTool[4853]: (CHIP) PersistentStorageDelegate Set Key StartKeyID
2022-02-18 12:52:01.170967-0800  localhost CHIPTool[4853]: (CoreFoundation) [com.apple.defaults:User Defaults] CFPrefsPlistSource<0x282562000> (Domain: com.apple.chiptool, User: kCFPreferencesCurrentUser, ByHost: No, Container: (null), Contents Need Refresh: No) skipping setting already-present value for key StartKeyID
2022-02-18 12:52:01.171063-0800  localhost CHIPTool[4853]: (CHIP) DevicePairingDelegate Pairing complete. Status ../../../../../../../../../../../CHIP-TOOL/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:846: CHIP Error 0x00000038: Invalid PASE parameter
2022-02-18 12:52:01.171277-0800  localhost CHIPTool[4853]: (CHIP) [com.zigbee.chip:all] ð¢ [1645217521171] [4853:261377] CHIP: [EM] Sending Standalone Ack for MessageCounter:389034827 on exchange 53941i

[Chiptool1-logs.txt](https://github.com/project-chip/connectedhomeip/files/8100368/Chiptool1-logs.txt)
[Failed-during-pase-session-setup-chiptool2-logs.txt](https://github.com/project-chip/connectedhomeip/files/8100369/Failed-during-pase-session-setup-chiptool2-logs.txt)
[Failed-during-pase-session-setup-M5-logs.txt](https://github.com/project-chip/connectedhomeip/files/8100370/Failed-during-pase-session-setup-M5-logs.txt)

bzbarsky-apple, 2022-02-19:
@kean-apple @vinay-apple That's all with the iOS chip-tool?  Because this:
```
./out/debug/standalone/chip-tool administratorcommissioning open-basic-commissioning-window 600 17 0 --timedInteractionTimeoutMs 10000

./out/debug/standalone/chip-tool pairing onnetwork 18 20202021 --commissioner-name beta
```
works fine for me on tip....

kean-apple, 2022-02-21:
I was able to verify on https://github.com/project-chip/connectedhomeip/commit/4394fcb17a8c5ae4209842f94e7acce5540393ba that I was able to pair with 2nd instance of iOS chiptool. 

With https://github.com/project-chip/connectedhomeip/commit/8a5032db871dfe4cb1650796f41dbf756ed5670e I was seeing a separate issue with my M5 board not able to connect to my Wifi on the 1st pairing so I haven't been able to validate the 2nd chiptool scenario yet

This is all with iOS chiptool . .Chaitanya mention chiptool CLI worked for him with RasPi as well

kean-apple, 2022-02-21:
OK. I confirmed with 8a5032db871dfe4cb1650796f41dbf756ed5670e the failure starts happening.

emargolis, 2022-02-22:
In the `src/controller/CHIPDeviceController.cpp` file (line 854) there is TODO. I believe this should be the root cause of the issue. For the ECM case the the correct PasscodeId (instead of `kDefaultCommissioningPasscodeId`) should be passed  to the `Pair()` function.

kean-apple, 2022-02-23:
The temp fix seems to resolve the issue. I'm able to pair with 2nd iOS chiptool instance

emargolis, 2022-03-25:
This was initially addressed with #15430 and then proper fix was implemented as part of work on #15735

closed 2022-03-25 by emargolis