project-chip/connectedhomeip #15374: Revoke commissioning does not work on master

cabin15, 2022-02-20:
#### Problem
Tested on 5cb8800a7f46bd7e9f62d32499c4aa053b0884fd (master branch)

Revoke commissioning does not work.

Tested between
- chip-tool
- chip-all-clusters-app

Precondition
- commissioning is already done.
- BCM commissioning window is opened with 120 seconds. (./chip-tool  pairing open-commissioning-window 12345678 0 120 2000 3840)

Reproduce steps
- Try revoking the existing commissioning window which is still open (./chip-tool  administratorcommissioning revoke-commissioning 12345678 0), but failed with IM error code 0x06

Am I using the wrong command? or I'm missing something?
This just works with test event 7.5 master branch whose checksum is b0af6ba42b6ce494748f18ecf3e054a400b8f45b

Here is the chip-tool and all-clusters-app log.

**chip-tool log**
sungmokshin@sungmokshin-P425-KE1WK:~/master/connectedhomeip/out/host$ ./chip-tool  administratorcommissioning revoke-commissioning 12345678 0
[1645381349.897172][240374:240374] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-zjsj48)
[1645381349.897401][240374:240374] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1645381349.897426][240374:240374] CHIP:DL: NVS set: chip-counters/reboot-count = 6 (0x6)
[1645381349.897702][240374:240374] CHIP:DL: Got Ethernet interface: enp8s0
[1645381349.897897][240374:240374] CHIP:DL: Found the primary Ethernet interface:enp8s0
[1645381349.898107][240374:240374] CHIP:DL: Got WiFi interface: wlp7s0
[1645381349.898134][240374:240374] CHIP:DL: Failed to reset WiFi statistic counts
[1645381349.898155][240374:240374] CHIP:IN: UDP::Init bind&listen port=5542
[1645381349.898200][240374:240374] CHIP:IN: UDP::Init bound to port=5542
[1645381349.898208][240374:240374] CHIP:IN: UDP::Init bind&listen port=5542
[1645381349.898229][240374:240374] CHIP:IN: UDP::Init bound to port=5542
[1645381349.898236][240374:240374] CHIP:IN: TransportMgr initialized
[1645381349.898253][240374:240374] CHIP:DIS: Init fabric pairing table with server storage
[1645381349.899164][240374:240374] CHIP:IN: Loading certs from storage
[1645381349.899205][240374:240374] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1645381349.899213][240374:240374] CHIP:IN: 0x04, 0x72, 0x3f, 0x85, 0xfc, 0xab, 0xf0, 0x91, 
[1645381349.899218][240374:240374] CHIP:IN: 0x33, 0x2f, 0xe1, 0x68, 0xc7, 0x98, 0x3d, 0x3c, 
[1645381349.899226][240374:240374] CHIP:IN: 0x0c, 0x68, 0x95, 0x28, 0x77, 0x13, 0x49, 0x1f, 
[1645381349.899235][240374:240374] CHIP:IN: 0xf5, 0xfb, 0x35, 0x0d, 0x7f, 0x58, 0x72, 0xa2, 
[1645381349.899244][240374:240374] CHIP:IN: 0x3d, 0x7c, 0x76, 0x00, 0xcd, 0x13, 0x07, 0x08, 
[1645381349.899253][240374:240374] CHIP:IN: 0xab, 0x69, 0x29, 0x8f, 0x6d, 0x0a, 0xee, 0xe3, 
[1645381349.899261][240374:240374] CHIP:IN: 0xcc, 0x96, 0x04, 0xd0, 0xfa, 0x34, 0xa7, 0xf3, 
[1645381349.899269][240374:240374] CHIP:IN: 0x35, 0x40, 0x1e, 0x6d, 0x8e, 0xe6, 0x5d, 0xdc, 
[1645381349.899275][240374:240374] CHIP:IN: 0x69, 
[1645381349.899301][240374:240374] CHIP:IN: Generated compressed fabric ID
[1645381349.899311][240374:240374] CHIP:IN: 0x60, 0xbb, 0x2f, 0x23, 0x0a, 0x09, 0xab, 0x5d, 
[1645381349.899436][240374:240374] CHIP:IN: Loading certs from storage
[1645381349.899469][240374:240374] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1645381349.899478][240374:240374] CHIP:IN: 0x04, 0x72, 0x3f, 0x85, 0xfc, 0xab, 0xf0, 0x91, 
[1645381349.899486][240374:240374] CHIP:IN: 0x33, 0x2f, 0xe1, 0x68, 0xc7, 0x98, 0x3d, 0x3c, 
[1645381349.899494][240374:240374] CHIP:IN: 0x0c, 0x68, 0x95, 0x28, 0x77, 0x13, 0x49, 0x1f, 
[1645381349.899502][240374:240374] CHIP:IN: 0xf5, 0xfb, 0x35, 0x0d, 0x7f, 0x58, 0x72, 0xa2, 
[1645381349.899509][240374:240374] CHIP:IN: 0x3d, 0x7c, 0x76, 0x00, 0xcd, 0x13, 0x07, 0x08, 
[1645381349.899516][240374:240374] CHIP:IN: 0xab, 0x69, 0x29, 0x8f, 0x6d, 0x0a, 0xee, 0xe3, 
[1645381349.899523][240374:240374] CHIP:IN: 0xcc, 0x96, 0x04, 0xd0, 0xfa, 0x34, 0xa7, 0xf3, 
[1645381349.899531][240374:240374] CHIP:IN: 0x35, 0x40, 0x1e, 0x6d, 0x8e, 0xe6, 0x5d, 0xdc, 
[1645381349.899537][240374:240374] CHIP:IN: 0x69, 
[1645381349.899554][240374:240374] CHIP:IN: Generated compressed fabric ID
[1645381349.899562][240374:240374] CHIP:IN: 0xba, 0x2f, 0x0b, 0xa3, 0x34, 0x77, 0x25, 0x90, 
[1645381349.899670][240374:240374] CHIP:IN: Loading certs from storage
[1645381349.899700][240374:240374] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1645381349.899709][240374:240374] CHIP:IN: 0x04, 0x72, 0x3f, 0x85, 0xfc, 0xab, 0xf0, 0x91, 
[1645381349.899717][240374:240374] CHIP:IN: 0x33, 0x2f, 0xe1, 0x68, 0xc7, 0x98, 0x3d, 0x3c, 
[1645381349.899724][240374:240374] CHIP:IN: 0x0c, 0x68, 0x95, 0x28, 0x77, 0x13, 0x49, 0x1f, 
[1645381349.899732][240374:240374] CHIP:IN: 0xf5, 0xfb, 0x35, 0x0d, 0x7f, 0x58, 0x72, 0xa2, 
[1645381349.899740][240374:240374] CHIP:IN: 0x3d, 0x7c, 0x76, 0x00, 0xcd, 0x13, 0x07, 0x08, 
[1645381349.899747][240374:240374] CHIP:IN: 0xab, 0x69, 0x29, 0x8f, 0x6d, 0x0a, 0xee, 0xe3, 
[... 339 lines cut ...]
[1645381349.917758][239997:239997] CHIP:DMG: 
[1645381349.917764][239997:239997] CHIP:DMG:    ],
[1645381349.917774][239997:239997] CHIP:DMG: 
[1645381349.917780][239997:239997] CHIP:DMG:    InteractionModelRevision = 1
[1645381349.917786][239997:239997] CHIP:DMG: },
[1645381349.917809][239997:239997] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_003C e=0 p=o
[1645381349.917828][239997:239997] CHIP:DMG: ICR moving to [AddingComm]
[1645381349.917837][239997:239997] CHIP:DMG: ICR moving to [AddedComma]
[1645381349.917847][239997:239997] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1645381349.917856][239997:239997] CHIP:EM: Piggybacking Ack for MessageCounter:4748825 on exchange: 5177r
[1645381349.917873][239997:239997] CHIP:IN: Prepared secure message 0x5634b51bd000 to 0x000000000001B669 (1)  of type 0x9 and protocolId (0, 1) on exchange 5177r with MessageCounter:15135933.
[1645381349.917886][239997:239997] CHIP:IN: Sending encrypted msg 0x5634b51bd000 with MessageCounter:15135933 to 0x000000000001B669 (1) at monotonic time: 1083647389 msec
[1645381349.917951][239997:239997] CHIP:DMG: ICR moving to [CommandSen]
[1645381349.917964][239997:239997] CHIP:DMG: ICR moving to [AwaitingDe]
[1645381349.918010][239997:239997] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:563207341 on exchange 5176r
[1645381349.918021][239997:239997] CHIP:EM: Found matching exchange: 5176r, Delegate: (nil)
[1645381349.918044][239997:239997] CHIP:EM: Rxd Ack; Removing MessageCounter:2405013545 from Retrans Table on exchange 5176r
[1645381349.918053][239997:239997] CHIP:EM: Removed CHIP MessageCounter:2405013545 from RetransTable on exchange 5176r
[1645381349.918596][239997:239997] CHIP:EM: Received message of type 0x10 with protocolId (0, 0) and MessageCounter:4748826 on exchange 5177r
[1645381349.918613][239997:239997] CHIP:EM: Found matching exchange: 5177r, Delegate: (nil)
[1645381349.918632][239997:239997] CHIP:EM: Rxd Ack; Removing MessageCounter:15135933 from Retrans Table on exchange 5177r
[1645381349.918670][239997:239997] CHIP:EM: Removed CHIP MessageCounter:15135933 from RetransTable on exchange 5177r

 #### Proposed Solution
<suggested fix, suggested enhancement>

sowmyassp, 2022-02-23:
@cabin15 - The status error 0xc6 says in spec says "NEEDS_TIMED_INTERACTION", So we have to pass timedInteractionTimeoutMs parameter along with other parameter.
./chip-tool administratorcommissioning revoke-commissioning 2 0 --timedInteractionTimeoutMs 1000

bzbarsky-apple, 2022-03-23:
Yep, this needs to be done as a timed invoke.

closed 2022-03-23 by bzbarsky-apple